prusa3d/PrusaSlicer #4082: Can't run prusa slicer at the command line

EagleRocker, 2020-04-16:
### Version
_Version of PrusaSlicer used goes here_
2.3
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-04-16:
Please provide some description, system info, etc.

EagleRocker, 2020-04-16:
I just updated my request. Thanks

lukasmatena, 2024-02-18:
Sorry for not getting back to you in time. The problem seems to be that you were trying to run `prusa-slicer-console`, which we only distribute for Windows. But I fully understand that you are probably no longer waiting for an answer.
Closing.

closed 2024-02-18 by lukasmatena