prusa3d/PrusaSlicer #4085: Slicer extremely slow to export gcode

jalfeld, 2020-04-17:
### Version
2.2.0+Win64

### Operating system type + version
Windows 10 Pro 64 Bit

### 3D printer brand / version + firmware version (if known)
Voron 2.2 running Klipper

### Behavior
The software is extremely slow to export gocde files. to export even the simplest of files takes on the order of minutes. In one test exporting the gcode for a 40mm x 40mm single layer (.2mm) print took over four minutes to complete. This is new behavior since upgrading to 2.2, previous versions  did not exhibit this issue. 

[0.2mm_layer_height.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4493341/0.2mm_layer_height.3mf.zip)

rtyr, 2020-04-17:
Extremely slow to export gcode files where?
I am unable to reproduce it with your 3mf, export is completed instantly (2.2.0+win64).

jalfeld, 2020-04-18:
It is extremely slow to export them to any location that I select. It does not matter if it my M.2 SSD, my secondary hard drive, or external media. I have been able to work around the problem by disconnecting my external media reader, this would seem to indicate that there is something strange in how Pruse Slicer is scanning drives in the background. I do not see why the software would need to scan for drives on all of the possible drive letters presented my a media reader when the file save location is not connected through the media reader (or USB for that matter).

rtyr, 2020-04-18:
Thank you for the clarification. Might be related to https://github.com/prusa3d/PrusaSlicer/issues/4071

jalfeld, 2020-04-18:
It could be, I discovered the workaround after posting the original issue. Really hoping a resolution is found for this, messing around with connecting and disconnecting my media reader every time I need to slice something is tedious....

bubnikv, 2020-10-17:
PrusaSlicer 2.3.0-alpha1 will be released soon. Please let me know whether the issue is fixed for you.

rtyr, 2021-02-16:
@jalfeld Is it ok with 2.3.0?

rtyr, 2021-03-22:
No response. Should be fixed since alpha1. Closing.

closed 2021-03-22 by rtyr