prusa3d/PrusaSlicer #4087: [Feature Request] Avoid crossing perimeters on all layers but first layer

Photogad, 2020-04-17:
### Behavior

Avoid crossing perimeters is a great option to reduce stringing with PETG. But it is a bad option also for the first layer because you need to cross perimeters on the first layer with PETG or else the PETG will accumulate on the nozzle and/or cause parts of the first layer to curl upwards.

If we can add an option (chekcbox) to disable "avoid crossing perimeters" only on the first layer, I think this will solve a lot of people's problems with getting a good first layer with PETG.

For the longest time I was struggling with getting my PETG to stick to the bed, and could not find the solution. Someone mentioned to me to disable "avoid crossing perimeters" which I had previously enabled, and all of my first layer problems went away, at the cost of more strings elsewhere on the print.

_Is this a new feature request?_
Yes

BeatSlayer, 2020-04-18:
I wanted exactly the opposite, an "Avoid crossing perimeters on first layer only" option.

Photogad, 2020-04-18:
> I wanted exactly the opposite, an "Avoid crossing perimeters on first layer only" option.

Well, why not both!

Curious why you want opposite?

BeatSlayer, 2020-04-18:
I want the opposite because when I print a model like a board with lots of small holes sometimes the nozzle will rip from the bed some of the hole's perimeters when traveling (300mm/s), I can set a slower travel, or lower the feedrate but that requested option would only take a click and not influence the rest of the print.

bubnikv, 2020-12-11:
I would be interested in real data.

also 

"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

Also the monotonic infill is now the default for bottom layers, thus it is being used on the 1st layer.

So do we still have a problem?

closed 2026-09-29 by prusaslicer-bot[bot]