prusa3d/PrusaSlicer #4088: Overlapping internal perimeters generated & "Detect thin walls" description misleading

rackley096794, 2020-04-17:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
* When printing cylindrical objects with varying thicknesses, internal perimeters are clearly generated in a way that they can create massive overlap.  This effectively results in extreme over-extrusion, leading to big printing artifacts.
* "Detect thin walls" effects this behavior.  With it turned OFF, the above problem state is the result.  When turned ON, the object is properly sliced and gap fill is used instead.

Massively overlapping internal perimeters with "Detect thin walls" turned off.  Note there are **FOUR** internal perimeters here (yellow color):
[image]

Correct use of gap fill with "Detect thin walls" turned on:
[image]

The practical result of massively overlapped internal perimeters pushes external perimeter extrusions outward:
[image]

### Suggested action
* Either fix this in slicer code or amend the tooltip text for "Detect thin walls" to indicate it can effect internal perimeter overlap.  Currently the descriptive text excludes this possibility:
[image]

_Is this a new feature request?_  No

#### Project File (.3MF) where problem occurs
[Overlapping internal perimeters issue.zip](https://github.com/prusa3d/PrusaSlicer/files/4494613/Overlapping.internal.perimeters.issue.zip)

Photogad, 2020-04-17:
What's the issue? Just leave thin walls on at all times.

rackley096794, 2020-04-17:
@Photogad Think bigger picture.  This issue was not filed for my benefit.  I know to turn it on for this object now.  This was filed to save others time, frustration and filament.  This setting controls something it does not indicate it controls.  From a user perspective the only way to logically get from the above broken state to a fixed state is to have "magic insider knowledge voodoo" or just start randomly checking and unchecking boxes.  This is not a user-friendly situation to say the least.

mathiasvr, 2020-05-05:
If the overlap is smaller `Detect thin walls` does not solve the issue and some over extrusion happens anyway. In my opinion, the slicer should have an option to disallow overlapping perimeters completely, to avoid over extrusion and make it easier to tune perimeter widths. Slic3r++ has this behavior by default and `Overlapping perimeters` must be enabled to get the PrusaSlicer behavior.

closed 2026-09-29 by prusaslicer-bot[bot]