prusa3d/PrusaSlicer #4089: Not possible to cancel an upload task to non-existing host

DominikPalo, 2020-04-17:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_
2.2.0+

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
macOS 10.15.4

### Behavior
* _Describe the problem_
It is not possible to cancel an upload task (while the status is uploading... although it is actually not uploading, just trying to connect to the host)
* _Steps needed to reproduce the problem_
  * Set the print host IP to a non-existing one
  * Try to upload a sliced model to the host
  * Go to the Print Host Upload Queue window - the upload is in "Uploading" status (would be better to show something like "Connecting..."
  * Try to cancel the problematic task
  * Nothing happens (after a ±minute of waiting, the task finally goes to the "Error" state

* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

See: [image]

JohnnyDeer, 2020-04-20:
Hi, unfortunately we´re not able to reproduce it, works how should. Can you please attach 3mf?

FidelCapo, 2021-05-03:
Do you still have this problem in last release?

rtyr, 2022-02-22:
No response. Closing.

closed 2022-02-22 by rtyr