prusa3d/PrusaSlicer #4092: Missing Layer ?

thierryzoller, 2020-04-18:
I stumbled upon this when gcode.ws visualised an empty layer; believing this was an gcode.ws bug I looked into the file. I don't know whether I am right or not, but I prefer reporting this finding at the risk of being wrong.
[Cube_GCODE_PLA_FINE_TEST.zip](https://github.com/prusa3d/PrusaSlicer/files/4496511/Cube_GCODE_PLA_FINE_TEST.zip)

### Version
Latest Build

### Behavior
* _GCODE produces a near empty layer when preview shows a full layer with infill.
* _Steps needed to reproduce the problem_
  * Open the File in a Text Editor - Go to line 779
* _Expected Results_
Gcode should draw what the preview shows.

* _Actual Results_
  * Skipped a layer

_Is this a new feature request?_
NO

[TESTCubeBUG.zip](https://github.com/prusa3d/PrusaSlicer/files/4496506/TESTCubeBUG.zip)

[Cube_GCODE_PLA_FINE_TEST.zip](https://github.com/prusa3d/PrusaSlicer/files/4496512/Cube_GCODE_PLA_FINE_TEST.zip)

rtyr, 2020-04-18:
Did you print it? What do you think is wrong at the line 779 and below?

thierryzoller, 2020-04-18:
It's an empty layer to my understanding

rtyr, 2020-04-18:
No it is not empty.

[image]

thierryzoller, 2020-04-18:
Semantics - it doesn't do what the preview shows.

thierryzoller, 2020-04-18:
;M900 K0.20
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.3

G1 E-1.50000 F2400.00000
G92 E0
G1 Z0.400 F15000.000
;AFTER_LAYER_CHANGE
;0.3

closed 2020-04-18 by thierryzoller

rtyr, 2020-04-18:
I guess you can see it now ;-).

thierryzoller, 2020-04-18:
Figured that Gscode.ws is full of unpatched bugs. It's a shame because it is the only one being able to show overlas and linewidth