prusa3d/PrusaSlicer #4093: [Feature Request] all settings located in sidebar next to the plate

robob3ar, 2020-04-18:
All print, filament and printer settings can be located on the sidebar next to the 3d view - because when you change something and want to see it sliced - you have to jump back, and so on.. it can take many clicks less if you have it in the sidebar.. I tend to do that a lot

Something that cura uses - just looking at it and I see no reason not to have it implemented like that ..

Also just trying out the cura octopi plugin - pretty great - even shows the temperatures of the printer and the camera output - I mean that's really impressive - I wander if this is something too much to expect from prusaslic3r 

- also, waiting for that neosanding/ironing feature, I really want to try that out - and I'm gonna try and use cura since it supports octoprint in such a nice way and ironing - slic3r has a much nice GUI now, but functionality of cura is what just impressed me - even tho I find it a bit clunky to work around.. 

_Is this a new feature request? Yes

rtyr, 2020-04-18:
Some news about ironing feature https://github.com/prusa3d/PrusaSlicer/issues/458#issuecomment-613348690

Personally, I don't like Cura GUI (especially with all settings enabled).

robob3ar, 2020-04-21:
Yeah Cura GUI is worse than prusaslic3r (slic3r was god awful before).. PS is much more cleaner - just has some features that are really great 
- I mean just leave the 3d view on the side, so when we change something in print settings we can see immediately (if auto slicing is enabled).. also that octoprint connect was amazing, you can abort print and see what's on the camera - but yeah cura's approach to listing everything without any distinction is a bit awful, lines of settings - which go really deep into detail - so much so, that I don't even wanna touch it

- but major downer with the cura - I couldn't get any good prints past a few mm, used the prusa settings for cura, but feels it's probably all worng - and that ironing feature really is awesome - I was printing guitar picks - and with ironing - no manudal sanding required, it's smooth on both sides .. anyay gonna check what's new on the ironing feature

wavexx, 2020-08-03:
I think a good approach could be to have the ability to detach the settings (or detach the plate view) into a separate window.

This way you could have both at the same time without actually changing the settings views.
In a multi-screen setup having the settings on the secondary screen would be _awesome_.

The Cura UI/layout is truly awful, and one of the main reasons I prefer slic3r/PS overall.

foreachthing, 2020-08-03:
My 2¢: I like the "new" Slic3r GUI, where you can choose which options you want to have quick access to. They're called favorites.
I wish PS would have those - I only change about three to four settings per different print. So ... it's getting boring quickly clicking through all these menus just to find this one setting (that was before search came to life. This new search is awesome, yet no substitute to favorites).

n8bot, 2020-08-03:
In the next version of PrusaSlicer, the settings window can be detached as its own separate window.

[image]
[image]

wavexx, 2020-08-03:
@n8bot I'm rebuilding the master version but didn't even notice this =)

Is there a shortcut to collapse the sidebar without having to use the menu/button? That's the only thing I'm missing.

n8bot, 2020-08-03:
In the preferences there are several modes, the third option allows the settings window to pop out.

[image]

bubnikv, 2020-10-16:
We now have floating presets dialog in PrusaSlicer 2.3.0-alpha1. Closing.

closed 2020-10-16 by bubnikv