prusa3d/PrusaSlicer #4096: Lift Z parameter is in 2 different locations

BirkBinnard, 2020-04-19:
### Version
2.2.0

### Operating system type + version
Window 10/64

### Behavior
The Lift Z parameter is in the Filament Settings/Filament Overrides window and in the Printer Settings/Extruder 1 window. This raises the question of which one actually determines the Lift Z function, and what happens if the 2 values are not equal.

_Is this a new feature request?_

Yes - Lift Z should be specified in only one place.

JohnOCFII, 2020-04-19:
The options in the "Override" area are only used if they exist, and "override (replace)" the other settings.  It is to allow different versions from the "standard" setting that is in the extruder area.

The ability to have overrides (again, which are only used when present) is a new and greatly appreciated feature.  Perhaps the term override is not clear in all languages?

BirkBinnard, 2020-04-20:
Aha! That meaning of "override" was not clear to me. Based on your explanation I can see how it could be beneficial in certain situations. Perhaps the hover help comment could be improved by adding a short explanation like you just provided.

OK to close this issue.

closed 2020-04-20 by BirkBinnard