prusa3d/PrusaSlicer #4097: [Bug] Gaps in Sliced file when printing lithophanes

nfons, 2020-04-20:
### Version
2.2.0 (latest)

### Operating system type + version
win 64

When printing lithophanes there are holes in the sliced file that does not exist when done with Cura.

Prusa slicer

[image]

[image]

3mf attached (different picture)
but bug still evident on the right side.

[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/4500573/bug.zip)

JohnnyDeer, 2020-04-20:
Hi, fixing model trough netfabb will solve it.

nfons, 2020-04-20:
@JohnnyDeer  these are from 3dprocks. havent had any issues when printing prior (even prusa guides tell to use that).

As well as the fact that it works fine on cura without issues.

if it was a model issue, i would expect it to be in cura as well

rtyr, 2020-04-22:
It is mainly a model issue, Cura just handled automatic repair better in this case. That is why we have "Fix through the Netfabb" option in PrusaSlicer (right click on the model). It will repair the model using Netfabb service.

nfons, 2020-04-23:
@rtyr im not so sure thats the case 100%. (could be wrong)

even after doing fix via netfabb:

[image]

Theres no indication that theres any hole there in the model

[image]

and in cura:

[image]

(note the color is for me speed)

rtyr, 2020-04-23:
That is not a hole. You can check from the other side :-).

nfons, 2020-04-23:
@rtyr  i dont think that is correct:

[image]

changing to speed color, you can clearly see its a gap.

Here is is w/ volumetric flow rate, and from the back:

[image]

rtyr, 2020-04-23:
Looking at your screenshots I have a question. Are you still commenting that same object you provided in 3MF file?

nfons, 2020-04-23:
@rtyr  sorry no, the 3mf was from a different. i think i mentioned earlier. but both are stls

rtyr, 2020-04-23:
Oh. So you are posting unrelated screenshots from completely different model. That's very helpful and not confusing at all :-). Well, if you have issue with different model, please provide 3MF.

nfons, 2020-04-24:
@rtyr 
[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/4530670/bug.zip)

rtyr, 2020-04-24:
These are not gaps, its where perimeters meet. The model is just too thin in some areas to fit two 0.45mm extrusions, so it is printed this way. Cura handles it differently, it will allow very big perimeter overlap. Anyway, I believe it will be printed fine.

[image]
[image]

closed 2020-04-30 by nfons