prusa3d/PrusaSlicer #4098: bridge infill and not top/bottom solid infill does not touch the perimeters

JSkrat, 2020-04-20:
### Version
2.1.1+win64

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Graber i3 + marlin

### Behavior
* Solid infill and bridges do not touch the most internal perimeter
* To reproduce just slice objects with 4 perimeters and 4 top/bottom layers
* Infill is expected to touch internal perimeter for structural integrity
* In fact there is a gap for no reason (pls see imgur images below)

#### Project File (.3MF) where problem occurs
[helmet_hook.zip](https://github.com/prusa3d/PrusaSlicer/files/4503081/helmet_hook.zip)

https://i.imgur.com/eK4uAS8.png
https://i.imgur.com/zcBdK42.png
https://i.imgur.com/D2bW4z2.png

JohnnyDeer, 2020-04-20:
Hi, this is solved in version 2.2.0, please try it.

rtyr, 2020-04-22:
Fixed in 2.2.0. Closing.

closed 2020-04-22 by rtyr