prusa3d/PrusaSlicer #4100: Heght Range -> Support Material doesn't work

3Dietrich, 2020-04-20:
### Version
2.2.0+

### Operating system type + version
MacOS Sierra, 10.12.6

### 3D printer brand / version + firmware version (if known)
mk3s 

### Behavior
_Describe the problem_
I want set Support to "everywhere" and block it by 'Hight range' from 10...top.
I know there is a [bug](https://github.com/prusa3d/PrusaSlicer/issues/3060) but it is set to "fixed" and is closed but I found it today in 2.2.0+ and it's definitely there.

_Steps needed to reproduce the problem_
- add Hight Range
- (set it to from 10 to 33, but it doesn't work with any height setting)
- try to add modifier: /Add Settings/**Support Material**. 

_Expected Results_
I think it should add the possibility to enforce/block support (like the modifier do). Or wouId learn what it does, but..

_Actual Results_
I can click, but nothing happens.. 
 
_error message?_
no

_Screenshots?_
I made a screen capture (in attached .zip)

_Is this a new feature request?_
if this is not a bug then it's a feature request

#### Project File (.3MF) where problem occurs
[HeightRange_Support_dsn'twrk.zip](https://github.com/prusa3d/PrusaSlicer/files/4504172/HeightRange_Support_dsn.twrk.zip)

YuSanka, 2020-04-21:
Currently, we don't support "Support materials" setting overriding for parts/modifiers and layer height ranges. It is implemented just for objects. But we leave this issue open as a feature request for the voting.

3Dietrich, 2020-04-21:
I'm not sure I understood you.
What mean "currently"? In version 2.2.0+ it actually supports that function as if it would work. Gray it out if you really need the entry or better: kick the entry out.

BTW: The alternative: adding a Support Blocker over the object and move it upwards (I remember once that moving the object z didn't work but) that works..

bubnikv, 2020-10-09:
We do not support "support material" attributes at the layer ranges, therefore this option will be removed in PrusaSlicer 2.3.0-alpha1.
Closing.

closed 2020-10-09 by bubnikv