prusa3d/PrusaSlicer #4102: Enabling 'Avoid Crossing Perimeters' causes glitchy travel moves when printing support material.

Lazerlord10, 2020-04-20:
### Version
2.2.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Folger Tech FT-5 with Marlin Firmware

### Behavior
The travel moves between different support material towers are not as expected. The printhead travels seemingly at random across the printbed until eventually getting to the next tower to print. Here is a video showing this motion: [LINK](https://www.reddit.com/r/3Dprinting/comments/g4xyw0/prusaslicer_adds_unnecessary_and_weird_travel/)

This occurs when printing support materials when the 'Avoid crossing perimeters' setting is enabled. Disabling 'Avoid crossing perimeters' removes these weird travel moves.
[image]

The expected travel moves when printing supports is a travel move that goes directly from the end of the previous support tower to the next one so long as the object is not along the path of that travel move.

[image]

JohnnyDeer, 2020-04-21:
Hi, thank you for reporting, please provide 3mf file.

Lazerlord10, 2020-04-21:
Attached to main issue (didn't know I had to zip it first, sorry).

JohnnyDeer, 2020-04-22:
Its ok, issue confirmed, thank you.

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

[image]

closed 2020-12-11 by bubnikv