prusa3d/PrusaSlicer #4103: Super simple sphere with hole through the middle results in "empty layers detected, the output would not be printable" error

mjparme, 2020-04-21:
### Version

[image]

### Operating system type + version

MacOS Catalina 10.15.4

### 3D printer brand / version + firmware version (if known)

Custom build, Marlin 2.0.1

### Behavior

Slicing a super simple model I whipped up in OpenSCAD results in "empty layers detected, the output would not be printable". PrusaSlicer reports the model as manifold.

[image]

The OpenSCAD code:

```
$fn = 100;

difference() {
    sphere(d = 18.55);
    cylinder(d=8, h=20, center=true);
}
```

#### Project File (.3MF) where problem occurs

[ticTacToe3dPiece.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4507021/ticTacToe3dPiece.3mf.zip)

mjparme, 2020-04-21:
FWIW, Cura 4.5.0 slices this without issue.

rtyr, 2020-04-21:
Bottom of the model is too thin to be printed with your extrusion width, so it is expected behavior. You can use "Detect this walls" parameter to avoid it.

mjparme, 2020-04-21:
@rtyr I flattened out the bottom by differencing a cube so now there are 4 perimeters on the first layer and all is well (that will help with bed adhesion anyway, first one would have been lucky to stay).

Will just chalk this up to user error. Sorry about that!

closed 2020-04-21 by mjparme