prusa3d/PrusaSlicer #4104: 3DConnexion Settings CTRL+M not working

foreachthing, 2020-04-21:
### Version
version_2.2.0-157-g40e4be6ed

### Operating system type + version
Windows 10 x64

### Behavior
Maybe I missed something but:
* 3Dx-Device is really slow
* Press **CTRL+M** ..... does nothing.
  * also, it's not listed in the keyboard shortcuts list.

### Expected
* Press CTRL+M to change the 3Dx settings

bubnikv, 2020-04-21:
You have to have the 3D scene active. It has to have a keyboard focus.

enricoturri1966, 2020-04-21:
Newer versions of PrusaSlicer on Windows use the 3DConnexion device settings, that's why the settings dialog embedded into PrusaSlicer is no more available.

bubnikv, 2020-04-21:
>  Newer versions of PrusaSlicer on Windows use the 3DConnexion device
settings, that's why the settings dialog embedded into PrusaSlicer is no
more available.

That is the upcoming PrusaSlicer 2.3. PrusaSlicer 2.3 will not need the
built-in settings dialog.

foreachthing, 2020-04-21:
Ok.
But I'm still on 2.2 and my SpacePilot is ultra-slow. 3Dx speed settings are ineffective.
Do I have to wait for 2.3 now?

enricoturri1966, 2020-04-21:
@foreachthing 
You wrote "2.2.0-157-g40e4be6ed" as the version referring to this issue, so the guess was that you built PrusaSlicer by yourself.
That version already contains the changes I was talking about in my previous comment.

foreachthing, 2020-04-21:
Yes, I did build it myself.
What I'm saying is, that the settings in the 3Dx panel have no effect.
Speeds are all set to 100% and it's still ultra-slow.

Now I am at commit 9bc96bf2 - no change.

enricoturri1966, 2020-04-24:
@foreachthing
Could you please try to copy the attached files into:
C:\Users\<username>\AppData\Roaming\3Dconnexion\3DxWare\Cfg
to see if this helps in fixing this issue ?

https://www.dropbox.com/s/prziag60rutngap/prusa-slicer.xml?dl=0
https://www.dropbox.com/s/2vov4o86tf09439/prusa-slicer-console.xml?dl=0

foreachthing, 2020-04-27:
No, unfortunately I can't.

If I add this line `<Transport>RawInput</Transport>` to my xml, it gets immediately overwritten.
And for the console:
[image]

foreachthing, 2020-04-27:
Ok ... So, I tried this:
* remove old file
* restart PS
* open 3Dx and import new settings

Whatever I do, it always removes this `Transport` section!

What I import:
```
  <AppInfo>
    <Signature>
      <Name>PrusaSlicer</Name>
      <ExecutableName>prusa-slicer.exe</ExecutableName>
      <Transport>RawInput</Transport>
    </Signature>
    <Options />
  </AppInfo>
```

What it imports:
```
  <AppInfo>
    <Signature>
      <Name>PrusaSlicer</Name>
      
      
    <ExecutableName>prusa-slicer.exe</ExecutableName></Signature>
    <Options/>
  </AppInfo>
```

enricoturri1966, 2020-04-27:
Please try this:
1) close PS and 3Dx
2) Copy the attached file to replace those automatically generated by 3Dx
3) start PS and 3Dx

If you try to manually edit the files you need to ensure that:
line 1 -> standalone="no"
line 6 -> Default="true"
line 11 -> <Transport>RawInput</Transport>
are all set or 3Dx will override your changes (it is likely doing some validation in background)

foreachthing, 2020-04-27:
I didn't pay enough attention to details :-\
I fixed line 1 and 6 and it got better  - not great but better.
Right now I have everything (main speed and all the individual axis) at full speed and it's OK. Could be faster, though! ;-)

foreachthing, 2020-05-04:
It's just getting worse!
New build (https://github.com/prusa3d/PrusaSlicer/commit/28c0880b24f36e3b3c7c9fb8415725f32f1ae670), and I can't zoom anymore. It's jerky at a random direction.
I wish I could go back to your low-level hardware support (not through the driver)...

Every other application (SolidWorks, etc.) they work as normal.

foreachthing, 2020-05-26:
Ok ... I don't know if I should open a new issue for this:
My 3DX-devices work (poorly) in the "**3D editor view**" but not at all in "**Preview**"!
Can somebody please confirm (or deny!) this issue?

foreachthing, 2020-06-15:
This is not fair:
In PS 2.2 (latest release), my 3DConnexion device works perfectly!!
But in the latest commit from today (https://github.com/prusa3d/PrusaSlicer/commit/76d9f0f5091250f8ca2002fdfeb94e4d03c615f0) it does not work at all. All the axis are reversed, for one. Then bringing up the 3DX-Menu and reversing all the axis - does nothing!
Speed is just ... not what it should be.
I am very sad! :'-(
It worked in PS 2.2 just the way _I_ liked it and now it doesn't....

bubnikv, 2020-10-09:
@foreachthing 
Would you please test with the current master? We are close to the first alpha.
If it is not working for you as expected, would you please give us an exact type of your device? Thank you.

foreachthing, 2020-10-13:
So - finally was able to rebuild the latest commit (#f0de7f1). But still no banana!
When downloading the prebuild zip, all is just fine. That's with the CTRL+M still available.

Lastest Build: axis are all wrong, speeds are wrong.... can't find anything I'd like - sorry :-(
"Fit" does not zoom to view, are any other "view" buttons are working for that matter.

Device: 3DConnexion SpacePilot Pro (FW: 4.11)
3DxWare 10.6.3
3DxWinCore 17.6.3.16588

<details>
<summary>Hardware Info</summary>
Device instance path
<pre>HID\VID_046D&PID_C629&MI_01\9&75C0191&0&0000</pre>

Hardware IDs:
<pre>
HID\VID_046D&PID_C629&REV_0411&MI_01
HID\VID_046D&PID_C629&MI_01
HID\VID_046D&UP:0001_U:0008
HID_DEVICE_UP:0001_U:0008
HID_DEVICE
</pre>
</details>

If you need more info, I'd be glad to help!

bubnikv, 2020-10-13:
Ctrl-M is no more available on Windows with the current master.

We have received a great news from 3DConnexion that they added configuration for prusa-slicer.exe into their 3DConnexion driver package version 10.7.0. This package shall install
C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\prusa-slicer.xml

foreachthing, 2020-10-13:
Unfortunately has the SpacePilot Pro "expired" with this driver version :-(
But, maybe it might still work. I'd be glad to try, though!! 👍

foreachthing, 2020-10-15:
Ok, I've installed the latest driver from 3DConnexion.
My SpacePilot is now a "discontinued device" 👍 

Anyways! It works much better now.

BUT: It works in the Editor View just as expected.
In Preview ... not so much! If I don't touch my mouse (2D mouse!!) while orbiting with my SpacePilot, all movements are junky (< 1 fps). So, now I move my 2D-mouse while orbiting, it's all good! What is going on here?? The faster I move my 2D-mouse, the more responsive my SpacePilot gets.

Need more info? Please let me know.

bubnikv, 2020-10-15:
The G-code viewer application has to have its own xml file installed 
C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\prusa-gcodeviewer.xml
just take the prusa-slicer.xml and replace the prusa-slicer.exe with prusa-gcodeviewer.exe

we have fixed some issue yesterday which prevented prusa-gocdeviewer.exe to work with the 3D mouse.

foreachthing, 2020-10-15:
Ok, to clarify:

1. Exit PrusaSlicer
1. Stop 3dx driver
1. COPY `prusa-slicer.xml` to `prusa-gcodeviewer.xml`
1. Rename string `<ExecutableName>prusa-slicer.exe</ExecutableName>` to `<ExecutableName>prusa-gcodeviewer.exe</ExecutableName>`
-- Also tried to add that file in `c:\Users\USER\AppData\Roaming\3Dconnexion\3DxWare\Cfg\`
1. Start 3dx driver
1. start PrusaSlicer

Aaaand that does not work. Or, same behavior as before.
Something else to try?

bubnikv, 2020-10-15:
>    2. Stop 3dx driver
>    3. COPY prusa-slicer.xml to prusa-gcodeviewer.xml
>    4. Rename string <ExecutableName>prusa-slicer.exe</ExecutableName> to
>    5. Start 3dx driver
>    6. start PrusaSlicer

recompile from the current master.

foreachthing, 2020-10-15:
I'm at commit https://github.com/prusa3d/PrusaSlicer/commit/3fc12fdaaad6a2702790dabe4f21b5dd6bf221f1 but I still need to wiggle my mouse to get it working properly (in preview).
Maybe it's because my device is no longer supported???

BTW: the speed settings are not saved. Who does the saving, 3DConnexion or PS?

bubnikv, 2020-10-15:
> I'm at commit 3fc12fd

3DConnexion should save the settings.

The 3DConnexion shall detect the application, see PrusaSlicer at the top.
Sometimes it shows Explorer, which is not correct.

[image: image.png]

foreachthing, 2020-10-15:
I've been using 3DX-devices since 2004 ;-) I use it every day at work and even got one at home.
So .. here's my screenshot:
[image]

I wish that most of those commands would be standard in PrusaSlicer.
I had to create a Macro for each button. Ever for top, left, right and ISO view ... etc.
Every button assignment with (PS) in it, is a custom macro.

bubnikv, 2020-10-15:
@foreachthing 

There are too many of these devices around, we cannot test and configure all of them.
We are in touch with 3DConnexion. You may pass your configuration of PrusaSlicer/3DConnexion here, I may pass it to them, so they will integrate it into their driver installer.

bubnikv, 2020-11-28:
Could you please retest with the upcoming PrusaSlicer 2.3.0-beta1? Thanks.

foreachthing, 2020-11-28:
@bubnikv - THANK YOU!!!!! :heart_eyes:
It works perfectly!!!!

bubnikv, 2020-11-28:
A closed issue is a good issue.
Thanks @enricoturri1966

closed 2020-11-28 by bubnikv