prusa3d/PrusaSlicer #4105: What version should I have?

beetlegigg, 2020-04-21:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

beetlegigg, 2020-04-21:
I downloaded as is shown as 'DRIVERS & APPS 2.3.0' (mac dmg file).  I installed it, choosing to remove the existing one, but the top bar of the slicer window shows 'PrusaSlicer-2.2.0+-2002003211132 based on Slic3r'  - Is this correct? What happend to 2.3?  I downloaded and repeated this twice to make sure I had not made a mistake.  
I then noticed that to download page says 
CHANGES IN 2.3.0
  New PrusaSlicer 2.2.0 (video about new features)
So I'm confused 2.3 or 2.2.  If it should be 2.3 where can I download it?

beetlegigg, 2020-04-21:
The download was for a MK3S printer, and Im running on macos 10.15.4

rtyr, 2020-04-21:
Yes, 2.2.0 is correct version of the PrusaSlicer. 2.3.0 is version of the drivers&apps package. Closing.

closed 2020-04-21 by rtyr