prusa3d/PrusaSlicer #4107: [FR] Machine limits turn on / off by a radio button

AndyWo83, 2020-04-21:
### Version
V2.2.0

### Operating system type + version
Win 10 Pro

### 3D printer brand / version + firmware version (if known)
Ender 3 - SKR V1.3 - TMC2130 - BMG Direct Drive Extruder with V6 Hotend

### Behavior
It would be possible to turn the "machine limits" function on or off using a radio button?
The main problem is that i want to set the normal acceleration to 800 and the travel to 2000.
M204 P and T are overwritten by "Maximum acceleration when extruding" at the same time and there are not separated.

rtyr, 2020-04-21:
Switch the gcode flavor in slicer to RepRap, it will disable machine limits.

PigeonFX, 2020-08-25:
I second this feature request.

rtyr, 2020-10-20:
Implemented in 2.3.0-alpha1. Closing.

[image]

closed 2020-10-20 by rtyr