prusa3d/PrusaSlicer #4108: Issues with "Empty layers detected" error when using support blockers

Guper92, 2020-04-21:
### Version
Version: 2.2.0+win64
Build: PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Windows 10 Home
Version 1903

### 3D printer brand / version + firmware version (if known)
Prusa I3Mk3S

### Behavior
I repeatedly get an error when slicing stating:

"Empty layers detected, the output would not be printable.
Object name: 85-45mm_NFA_Billet_C9-C4-C5-C45_All_V7.stl
Print z: 41.650000
This is usually caused by negligibly small extrusions or by a faulty model. Try to repair the model or change its orientation on the bed."
 
I can only get this error to go away by either moving one of my support blockers or deleting one of the parts on the build plate. The part I delete, however, isn't nearly 41.65mm tall and should have no affect on an "empty layers" error at 41.65mm. The support blocker that I can move to get rid of this error as well is not even touching the part that, when deleted, causes the error to go away. I have had this error MANY times now with different, but similar, models. With all of the models that I have experienced these issues with, none of them have an issue slicing when I do not use support blockers.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

#### Project File (.3MF) where problem occurs

[NFA-C9_PrusaSlicer-Issues.zip]

rtyr, 2020-04-21:
Please provide 3MF project file.

Guper92, 2020-04-22:
> Please provide 3MF project file.

I uploaded the zip file containing the .3MF file at end of the document right under the section titled "Project File (.3MF) where problem occurs)

rtyr, 2020-04-22:
It is not there, that is why asked for it :-).

Guper92, 2020-04-22:
> It is not there, that is why I asked for it :-).

For some reason something went wrong in the upload. It placed the tagline, but not the link to the upload even though I drag and dropped the file. Should be a working link now.

[image]

rtyr, 2020-04-22:
It is OK now, thank you. I managed to reproduce the issue.

Guper92, 2020-04-26:
OK Does that mean it's been fixed or just that it's acknowledged that it's a problem?

Guper92, 2020-05-11:
> It is OK now, thank you. I managed to reproduce the issue.

Any progress on this?

rtyr, 2020-05-11:
> 
> 
> OK Does that mean it's been fixed or just that it's acknowledged that it's a problem?

It means, that I was able to reproduce behavior you are describing. Now it is up to core PS devs to determine what to do about it and if it has priority over many other things they have on their plate.

lukasmatena, 2020-05-11:
Thanks for the report. I managed to reproduce the problem in 2.2.0, but not in current master. It appears it was flushed along with something else. However, this one would be worth testing again when 2.3.0 release gets close, that part of the code will still be touched.

Guper92, 2020-05-12:
> Thanks for the report. I managed to reproduce the problem in 2.2.0, but not in current master. It appears it was flushed along with something else. However, this one would be worth testing again when 2.3.0 release gets close, that part of the code will still be touched.

Awesome! I'll be sure to test this out and report back if there's an issue once I update to 2.3.0

cfelicio, 2020-05-21:
If not fixed, in the meantime I have a version published with this check disabled (as I was having issues printing stacked PPE and getting the same error): https://github.com/prusa3d/PrusaSlicer/issues/4062

FidelCapo, 2020-10-15:
This issue should be fixed in next version. We changed notifications for "Empty layers detected" errors and you can decide if you want to ignore them. There is still error notification with attached 3mf file, but I think it is ok, because one layer really have small extrusions made by supports generator.
I will close this issue for now. If there will be any problems related to this in next release, please feel free to reopen this issue. Thank you.

closed 2020-10-15 by FidelCapo