prusa3d/PrusaSlicer #4109: [Feature Request] XY Comp settings per filament

Makhaira, 2020-04-21:
Ok, so no problem to report here, just a suggestion.

**Scenario**
User prints different materials, and has different build plates. Due to different materials they adjust Live Z through G Code (or manually) to get the correct Z offset **but also the correct level of squish for adhesion on the first layer**. This differing level of squish leads to a requirement for different levels of XY Compensation/Elephant Foot Compensation.

**Suggestion**
On the assumption that users will only be changing build plates and/or Z offset for different filament types, XY Compensation should be moved to the FIlament options section, allowing for different settings per material.

I hope that's clear and logical enough for everyone, happy to talk through.

TD5023, 2020-04-25:
I was considering posting this, so I'm glad to see someone who thinks similarly. I'd add that bridge flow ratio should be in the filament section, too (mainly for the sake of flexible filaments). These would be excellent candidates to add to the filament override section so that it's more than just modifying retraction.

rtyr, 2021-02-16:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3059. Closing.

closed 2021-02-16 by rtyr