prusa3d/PrusaSlicer #4110: Thin walls and single perimeters

Phil1988, 2020-04-22:
### Version
2.2.0+win64

### Operating system type + version
Win7 x64

### 3D printer brand / version + firmware version (if known)
Self build 3D printer, Marlin Marlin-2.0.5.1

### Behavior
When slicing models, the PrusaSlicer always wants 2 outer perimeters or wont connect them to the rest of the part.
See here:
[image]

The shells around the hole and the walls on the bottom are placed seperately. This results to weak points on these already small walls -> they are already not or rarely connected to the real printed part.

I am printing with "detect thin walls" enabled
[image]

The wall itself is 0.47mm thick
[image]

Permimeter width is set below 0.47
[image]

I have to set the perimeter width all the way down to 0.24mm to have them connected:
[image]

Of course this is not printable with a 0.4mm nozzle (with  good results) and will also either massively increase the printing time for the rest of the model (if you want to maintain the same shell thickness) or make the overall object weaker and increase the printing time a bit (if you keep the numbes of perimeters the same, with the smaller perimeters width)

Expected behaviour would be to lay down a single layer that is really connected to the rest and printed in one movement like cura does.

A short test with cura shows, that it slices the thin walls correctly with for example 0.44mm set:
[image]
(and also prints the same way.... )

I really like the Prusaslicer because the print times are shorter and the result is almost the same compared to cura.

But some models cant be sliced with this slicer because some problems cant be solved by the user (eg. the problem descriped here or the "avoid crossing perimeters" which also currently doesnt work as it should).

rtyr, 2020-04-22:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/227, https://github.com/prusa3d/PrusaSlicer/issues/340, https://github.com/prusa3d/PrusaSlicer/issues/149, https://github.com/prusa3d/PrusaSlicer/issues/3611

I believe disabling the "Detect thin walls" parameter will help in this case.

marcosscriven, 2020-07-30:
Just reporting here I faced the same issue - reported in the forum https://forum.prusaprinters.org/forum/prusaslicer/bug-in-prusaslicer-its-disconnecting-a-bridge-from-perimeter/#post-237391

But to summarise here, here's the output with 'thin wall detection' on:

[image]

And here it is without:
[image]

You can see in the former case it's attempted to create four extra lines - but as the first one is essentially a small bridge (~12mm), and is disconnected from the rest of the perimeter, the printer ends up creating a blob here. It was enough to solidify and knock the nozzle, causing a print-failing layer shift.

marcosscriven, 2020-07-30:
@rtyr - you mention several similar issues:

1. You're closing such issues - are you saying this is the intended behaviour?
2. If it's not the intended behaviour, can we have an issue we can track where it will be fixed?
3. If it is as intended, can a warning be put in place in the UI that this might happen please?

rtyr, 2020-07-30:
I only closed some of them as a duplicate. It is a known behavior. and it's not as simple as one might think, @bubnikv is working on improved perimeter generator, which will hopefully fix some issues.

closed 2026-09-29 by prusaslicer-bot[bot]

marcosscriven, 2026-09-29:
Sigh. What is even the point of spending time giving you all that information only for you to throw it away.