prusa3d/PrusaSlicer #4112: Add thumbnails in settings GUI

Salamandar, 2020-04-22:
I just discovered PrusaSlicer can export thumbnails since 2.1.1… But I can't find the settings anywhere in the GUI. 
I can edit the printer .ini file by hand, and the thumbnail is correctly exported, but that's not really a "user-friendly" way ;)

Misfi, 2020-04-22:
Hi @Salamandar, could you please elaborate? What's wrong with the thumbnail export? 
Please provide all the information that you can when creating a [new issue](https://github.com/prusa3d/PrusaSlicer/issues/new). Pictures will also surely help.

JohnnyDeer, 2020-04-22:
Hi, more information not needed, thumbnails aren't in settings GUI right now. Let's make this the feature request.

thegarbz, 2020-04-22:
More info: Thumbnails are currently dependent on the Printer capabilities and printer settings. Printers like the MINI will cause PrusaSlicer to output thumbnails into the gcode which makes sense since the printer can display it. Printers like the MK3S this *didn't* make sense in the past as they can't display it on the screen. 

Software like OctoPrint and Octodash can display these thumbnails while sorting through gcode files or even while printing them.

Currently to make use of this functionality you need to define a new printer and edit the ini file manually or edit PrusaResearch.ini directly and add the thumbnail directive. 

This would be best to have as a setting.

Salamandar, 2020-04-23:
Yeah, I guessed this is a feature activated by printer name or whatnot. I wanted to try editing the Printer Notes to add other `PRINTER_*` and `PRINTER_VENDOR_*` to see if that changed anything but I was too lazy to search which "keywords" are enabled when using an "official" Prusa printer.

thegarbz, 2020-04-23:
@Salamandar It's not hard. Open up %appdata%\PrusaSlicer\vendor\PrusaResearch.ini

Look for [printer: model] in my case it's [printer:Original Prusa i3 MK3S]
add "thumbnails = 16x16,220x124"
save and done.

Salamandar, 2020-04-23:
@thegarbz Never said it was hard. I even said I did it. I said it's not user friendly. If you want to edit config files by hand, just call PrusaSlicer in command line and non-interactive mode ;)

ruedli, 2020-04-27:
I fully agree, will be more user friendly for sure, I opened a similar issue halve a year ago, but it was closed - pending proof of implementations. See #3488 . Since the "proof is there now" I asked to reopen it or create a new issue, or: (the best) implement it in the settings on e.g. the "general" tab. Hope it will get done, it does not sound to difficult, given the implementation of the feature is already there!

Salamandar, 2020-04-27:
I'll try and do a PR. As @ruedli says, that'd be a great "newby PR" for me.

Salamandar, 2020-04-28:
Well, that's more difficult than I thought. Whatever I do, i get 
```
terminate called without an active exception
[1]    19597 abort (core dumped)  ./src/prusa-slicer
```

I tried to add 
```
        optgroup->append_single_option_line("thumbnails", 0);
```
in Tab.cpp:1984.

Salamandar, 2020-04-28:
Well, Mouse3dController doesn't have a destructor. That's not normal.

lukasmatena, 2020-04-28:
> Well, Mouse3dController doesn't have a destructor. That's not normal.

Could you please be more constructive and tell us what the problem in the code is (if you can reproduce it on `master` branch)? Thanks.

Salamandar, 2020-04-28:
@lukasmatena I'm going to do a PR first, that's not the correct place for such discussion ;)

Kgoon1951, 2020-05-08:
> @Salamandar It's not hard. Open up %appdata%\PrusaSlicer\vendor\PrusaResearch.ini
> 
> Look for [printer: model] in my case it's [printer:Original Prusa i3 MK3S]
> add "thumbnails = 16x16,220x124"
> save and done.
> 
> It's just a pain that any update that Prusa issue will clobber these changes since on an update they are re-written.

Thanks for your more detailed instructions compared to what was given on the plugin site. Finally got it to work after following our instructions! Much appreciated, Thanks!

jneilliii, 2020-05-14:
> It's just a pain that any update that Prusa issue will clobber these changes since on an update they are re-written.

Exactly the reason the plugin page doesn't have the instructions written out like that. Every time there was an update it would break your settings. Creating a copy of the default profile in Prusa and modifying the custom one prevents that from happening. Really this should be a setting now that we've proven not just the Mini has the capability but any printer that is paired with OctoPrint can as well.

Kgoon1951, 2020-05-14:
Thanks for the clarification.

FixxCZ, 2020-06-01:
Adding my vote to enable this by default and make the option visible in the Settings.

ciaranmcnulty, 2020-06-10:
Yep I'd love to have this for my mk2.5S; I agree it should be off by default for these printers but enable-able through the GUI

thegarbz, 2020-06-11:
@ciaranmcnulty Why off by default? These printers ignore any Gcode starting with a preceding ; and there is already discussion of using M26 so the SD card doesn't even parse the selected code. Why would you disable it by default?

Salamandar, 2020-06-11:
@thegarbz because the generated gcode files are bigger.

Well, I tried to implement something but I couldn't get any code to run, and I don't have enough time on my hands for the moment :(

ciaranmcnulty, 2020-06-11:
Yep just for size

thegarbz, 2020-06-12:
I don't see how 50KB in what is usually a multi-megabyte file is even remotely relevant enough to justify making something off by default, and that's 50KB with with a 256x256 thumbnail, you could always default to 128x128.

Just for fun I checked the smallest thing I've ever printed and it was 40KB. I can't remember what it was because the filename wasn't descriptive enough "spacer" and it didn't have a thumbnail (ironically that would help here). But looking at the gcode it appears at somepoint I printed 4 tiny squares 0.5mm high. As a user I could not care less if that file was 90KB. Actually in this case it would have outright helped me. 

The next smallest file I have is 178KB, also sans thumbnail, no idea what "clip" was. 

So personally I do not remotely see size as a justification for this to be disabled by default.

Last night's print was 23.8MB. I guess 23.75MB if it didn't have a thumbnail.

ruedli, 2020-06-13:
Yes, and even more helpful is the preview and show as an icon in Windows
explorer (and Finder?), like for stl already. Super convenient...

shaiss, 2020-06-13:
> I don't see how 50KB in what is usually a multi-megabyte file is even remotely relevant enough to justify making something off by default, and that's 50KB with with a 256x256 thumbnail, you could [...]

This so we'll said and a great point.  100% agree!

livingstonjjks, 2020-06-28:
I agree, having the ability to turn on thumbnails as part of the default profiles would be fantastic

IARI, 2020-07-28:
I'd love this for my MK3s.

neophyl, 2020-08-24:
Now available in the latest version of Super Slicer. There’s options under the printer settings for off and 2 different thumbnail sizes, so once you turn it on and save it for that printer it gets output.

jneilliii, 2020-08-24:
Thanks @neophyl I'll be updating my repo and official plugin repository listing with links to SuperSlicer.

gdombiak, 2020-08-25:
Who is SuperSlicer? Is is a fork of PrusaSlicer?

gdombiak, 2020-08-25:
Just found it! Nicely done! Is the intention to stay as a fork and keep merging things from PrusaSlicer as needed while evolving on its own?

neophyl, 2020-08-25:
You would have to confirm with Merrill but from comments I have read then yes the plan is to continue building it against the current release of Prusa Slicer. So eventually the 2.3 will make its way in when released.

steve220, 2020-09-22:
Dumb noob question. If a copy of the base profile is created and modified to add the thumbnail previews, does one have to create another copy each time the base profile is updated by Prusa or an update to Prusaslicer is released?

mneumark, 2020-10-31:
Agreed. Would really appreciate this.

bubnikv, 2020-12-12:
Implemented in PrusaSlicer 2.3.0-beta3.

closed 2020-12-12 by bubnikv

mneumark, 2020-12-18:
Why does this seem to be missing from rc1? Also rc1 doesn't have MK3S+ references. Did something get messed up?

ruedli, 2020-12-19:
Under the tab "Printer settings" , section "firmware" you find "GCODE
Thumbnails". Specify the resolution(s) you want, like 25x25, 256x256. This
would generate two thumbnails, of 25x25 and 256x256 respectively.

jimb15122, 2022-11-07:
UvTools version 3.8.2 provides a workaround that can set the thumbnail to any specified layer or to the heatmap (avg pixel shade of all layers).  This is available from the CLI version also.  e.g. to set the thumbnail to layer 50, use:

UvToolsCmd.exe set-thumbnail my.sl1 50

See UvTools issue # 599 for details.