prusa3d/PrusaSlicer #4113: Error in Spiralize mode if doing a wipe on retract in filament

sbaeder, 2020-04-22:
### Version
2.2.0+win64
### Operating system type + version
Windows 10 64 bit 1909
### 3D printer brand / version + firmware version (if known)
My own build of V-Baby (CoreXY)
### Behavior
Trying to print spiral mode of a cube.  I do have the retract on layer change ticked on the filament as well as wipe on retract.  This seems to be cause of the issue.  When not doing the wipe, it works as expected.

* _Actual Results_
Bad slicing causing strange diagonal lines (due to the wipe on the retract?)
[image]

#### Project File (.3MF) where problem occurs
[303010 Calibration Cube.zip](https://github.com/prusa3d/PrusaSlicer/files/4517035/303010.Calibration.Cube.zip)

sbaeder, 2020-04-22:
Unchecking either the retract on layer change OR the wipe on retract will correct the issue.

rtyr, 2020-04-22:
I was able to reproduce it, thank you. It seems it only happens when filament overrides are used.

sbaeder, 2020-04-22:
thanks - yes, this seems to be the cause of it.  Was aable to proceed by turning either one off.

sbaeder@aol.com
508-331-1530 (cell)

bubnikv, 2020-12-09:
also duplicate of #5004
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/4c3525053134ceea7a458d2aa459dbbe0a3149c9

closed 2020-12-09 by bubnikv