prusa3d/PrusaSlicer #4114: Bad interaction between height-range modifiers and box-based modifiers.

ealexeev, 2020-04-22:
### Version
2.2.0 win64

### Operating system type
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3S, 3.8.1

### Behavior
* Introduction of a height-range modifier affects box-based modifiers outside of the height range.

* https://www.youtube.com/watch?v=bCs3SRSX07s

* Height-range modifier should only affect the selected layers/heights.  Modifiers outside of this range should not be affected.

* _Actual Results_
TL;DR;  Box-based modifiers outside of the height-range are affected in multiple ways.
  1.  They are affected by the height range modifier -- in the example they slow printing speed.
  2.  Some of their settings are also modifier -- in the example the infill type gets changed.
See youtube video for details.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[DtM-v3_0-0.4mmN-0.31mmL-x1.zip](https://github.com/prusa3d/PrusaSlicer/files/4517523/DtM-v3_0-0.4mmN-0.31mmL-x1.zip)

JohnnyDeer, 2020-04-23:
Hi, issue reproduced and confirmed, thank you for reporting.

bubnikv, 2021-06-03:
Looks like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5831

It seems to be fixed with PrusaSlicer 2.4.0-alpha1.

[image]

[image]

closed 2021-06-03 by bubnikv