prusa3d/PrusaSlicer #4116: error in your mode

Slonicak, 2020-04-23:
### Version 2.2.0
### Win 7 64bit
[image]

[vase_0.3mm_PLA_MK3S_1h9m.zip](https://github.com/prusa3d/PrusaSlicer/files/4524678/vase_0.3mm_PLA_MK3S_1h9m.zip)
[image]

###I3MK3S

###Slicer is not working properly.

ochm, 2020-04-24:
It looks like overheated filament. I printed lot of vases and never see such problem. But yes vase mode have problems, for example unprintable holes in bottom of objects.

espr14, 2020-04-24:
The original model is not correct (has no volume). Still, after fixing (Netfabb, Meshmixer) the issue remains so it's not in the model. The slicing preview (in yellow color) is correct but the final view has the flaws. Also non-vase result does not have the issue.

rtyr, 2020-04-24:
> 
> 
> The original model is not correct (has no volume). Still, after fixing (Netfabb, Meshmixer) the issue remains so it's not in the model. The slicing preview (in yellow color) is correct but the final view has the flaws. Also non-vase result does not have the issue.

I can confirm that.

Slonicak, 2020-04-24:
If you look closely, you can see that Slicer creates a new edge from the bottom to the top.

n808, 2023-08-01:
Somehow this problem just got much worse with 2.6.0. I printed this vase with 2.5.x where I now in hindsight can see a slight outline of an extra edge, but with 2.6.0 it go so bad I considered the print a failure. I switched to Cura to get a good print. Comparison photos PrusaSlicer 2.6 vs Cura prints, screenshot of gcode viewer and 3mf files attached. 

[image]
[Low-poly_Rose_Vase.zip](https://github.com/prusa3d/PrusaSlicer/files/12233016/Low-poly_Rose_Vase.zip)
[image]

gmoissey, 2023-10-25:
Experiencing the exact same issue and it is very annoying. Bump.

gopgnz, 2023-11-01:
I am also experiencing this issue on 2.7.0 alpha1, printing using Prusa XL and 1.4mm nozzle... this strange seam is very noticeable 
[image]

zxcv-org, 2023-12-22:
I wrote the code linked below because I was hitting this issue and was similarly annoyed by the fudged "seam". The code gets rid of the fudged seam pictured above (for every model I've thrown at it so far), and also tries to avoid jumping across horizontal steps (commonly used to create intentional gaps in some models). It also preserves the non-vase portions of the print as-is, and passes through stuff like fan speed changes etc.

https://github.com/zxcv-org/gcode-respiralizer

FWIW, I believe this bug (#4116) is a dup of #2841 - there's more discussion and examples there.

devoh747, 2023-12-27:
A fix that came from Orca was made and is in here https://github.com/prusa3d/PrusaSlicer/issues/2841 as a patch.  Does anyone know how to apply this and compile it so we can test the fix?

lukasmatena, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1), thanks to @andrewboktor for the implementation. Closing.

closed 2024-03-21 by lukasmatena