prusa3d/PrusaSlicer #4117: Weird Vase Issues (Resolved)

erikmcclain, 2020-04-24:
### Version
2.2.0 Win64

If the setting Minimum Travel after retraction is set to on, you get some quirky behavior in vase mode.  Turn it off and it will slice normal.

[image]
[image]

[image]

erikmcclain, 2020-04-24:
actually it seems it can happen if retract on layer change is one as well
[image]

rtyr, 2020-04-24:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4113. Closing.

closed 2020-04-24 by rtyr