prusa3d/PrusaSlicer #4118: Slicing  produces output with missing and incomplete layers, but mirrored object works fine.

hm17gh1, 2020-04-24:
### Version
Version: 2.2.0+linux-x64
Build: PrusaSlicer-2.2.0+linux-x64-202003211856

### Operating system type + version
Operating System: Unix
System Architecture: 64 bit
System Version: Linux 4.15.0-96-generic x86_64

### 3D printer brand / version + firmware version (if known)
2 different custom versions

### Behavior
* _Describe the problem_
When slicing the attached STL, the output has many missing/incomplete layers followed but layers printed in mid-air. The stl is manifold:yes.
The same object works as expected in Cura, Slic3r, and older versions of PrusaSlicer.
Interesting fact: if the same object is mirrored, it works flawlessly. 

* _Steps needed to reproduce the problem_
Load stl, look in the produced layers

* _Expected Results_
correct slicing
* _Actual Results_
missing/incomplete layers

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[tubeJoiner.zip](https://github.com/prusa3d/PrusaSlicer/files/4526569/tubeJoiner.zip)

hm17gh1, 2020-04-24:
Tested this in older versions of PrusaSlicer:
PrusaSlicer-2.0.0+linux64-201905201652   is the last version that works well for this model
PrusaSlicer-2.1.0+linux64-201909160925 and all newer are manifesting the problem.
both cura,slic3r work as expected.

hm17gh1, 2020-04-24:
[Extruder-TubeJoiner.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4526985/Extruder-TubeJoiner.stl.zip)

foreachthing, 2020-04-24:
What if you mirror the mirrored?

hm17gh1, 2020-04-24:
> What if you mirror the mirrored?

tried that, but double mirrored object is wrong again.

rtyr, 2020-04-24:
I am unable to reproduce it on 2.2.0+win64 (screenshot). The model is not OK though, so you might try to fix it before slicing. It will probably solve the issue even on your system. It seems like similar situation to your other issue https://github.com/prusa3d/PrusaSlicer/issues/4066.

[image]
[image]

hm17gh1, 2020-04-24:
modelin stl is produced with openscad, and is working well on all other slicers i've tried, including older (PS2.0.0). I do not have windows, and prusaslicer should not depend on 3rd party windows software to "fix" the models.

rtyr, 2020-04-24:
Yes, you already mentioned this. That doesn't change the fact, that your model is clearly not OK and you can easily fix it using PrusaSlicer.

Mofef, 2020-04-25:
I had the exact same problem with the exact same setup (linux, openscad, prusa slicer 2.2)
When I googled for "prusaslicer repair model" (thx for the hint @rtyr) I found another issue: https://github.com/prusa3d/PrusaSlicer/issues/2974 The OP there says 

> Saving a 3mf project containing the model and reopening it in Prusa Slic3r, the problems are detected, repaired and the model is sliced correctly.

So I tried that and it worked for me. :) I assume this is a duplicate of #2974 then....

If I would understand the actual problem with the model better I would open an issue at openscad. Maybe someone can help with that?

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo