prusa3d/PrusaSlicer #4119: Save Octoprint API key (and other credentials) in separated config file

Salamandar, 2020-04-24:
My PrusaSlicer config is on Github. That's neat, but I have Octoprint configured.
So… my Octoprint API key is on github too. Not great.

Would it be possible to split credentials (such as octoprint IP address and API key) to another .ini file, that would be in my .gitignore ?

I'm sure I'm not the only one in this situation…

rtyr, 2021-02-16:
Credentials are now saved in different ini file (physical printer). Can we close this issue?

Salamandar, 2021-02-17:
@rtyr small question : let's say I git clone my prusaslicer config, without the physical printer file. Does PrusaSlicer crash because it can't load the file ? Or does it create a dummy file ? Or does it open the "new printer" prompt windows ?

Salamandar, 2021-02-17:
Well, I just tested : Nothing, no crash, no warning, no file creation. I like it. Closing.

closed 2021-02-17 by Salamandar