prusa3d/PrusaSlicer #4124: Total number of layers and printing time

Denethor85, 2020-04-25:
Hi,

I would need to insert in the personalized G-Code the total number of Layers of a print and the time to complete it.
I have successfully entered the Layer indication in print with
`;LAYER:[layer_num]`

But it would be useful to include in the initial G-Code also the total printing time and the total of the layers

Is there already any command or function to do this?
Or you can insert it in future releases?

Sorry for my english

MrSimmmons, 2022-08-14:
+1 would be cool. or better yet, the ability to see the total amount of layers in the UI maybe in the `Sliced Info` section at the bottom right

foreachthing, 2022-08-14:
Where do you want it? ;-)
This can easily be added with post processing, like with this: https://github.com/foreachthing/Slic3rPostProcessing/.

foreachthing, 2022-08-16:
I've just added an option to my spp script to add the total layer count.

before:
```
; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.35mm
; first layer extrusion width = 0.50mm

M107 ; disable fan
```

After:
```
; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.35mm
; first layer extrusion width = 0.50mm
; total number of layers: 64

M107 ; disable fan
```

spikeygg, 2023-07-08:
> Hi,
>
> I would need to insert in the personalized G-Code the total number of Layers of a print and the time to complete it. I have successfully entered the Layer indication in print with [...]

If you add this to one of your Start or End G-code areas you'll see the value in the comments:
[image]

You can populate a bunch of different values using the g-code variables / placeholders found here: https://help.prusa3d.com/article/list-of-placeholders_205643

Just need to use curly braces {} or square brackets [] to have it resolved into the value.

closed 2026-09-29 by prusaslicer-bot[bot]