prusa3d/PrusaSlicer #4126: How to get better printing results when using support?

moelski, 2020-04-26:
### Version
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,095MB
OpenGL installation
GL version:   4.6.0 - Build 27.20.100.7987
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 530
GLSL version: 4.60 - Build 27.20.100.7987

### 3D printer brand / version + firmware version (if known)
SparkCube XL
Firmware : Klipper 0.8 with OctoPrint on top

### Behavior
My results when using support in PS don´t look very good. Let me show you a sample first: 
[image]
As you can see the first layer on top of the support is not really good. The filament tracks are to small. The result are loose structures within the layer. 
I used a contact distance of 0.11 mm for this to print. But even if I lower or rise the value it doesn´t get better at all. 

So I started looking in PS what I could change to get better results. First I tried synchronize with layer option. But this results in a very sticky support structure. You can´t peal it off very well. 
Next I tried different contact distances. But again the results are not really better. It ends always with very sticky support or with a very ugly "first" layer over the support.

My next try was to check the volumetric flowrate. 
[image]
And if I use contact distances > 0 and no layer sync I can see that the normal Flow is around 3-4 mm³/s (Pointer 2). But the first layer after the support uses 13 (Pointer 1) ... That´s over 3 times more. For me this looks a little bit extrem and I don´t know if my hotend can handle this really well. 
So first question ... 
**How can I adjust the Flowrate for the brown lines in the preview picture shown above?** 

And a more general question ... 
**How can I adjust the support in a way that it is not so sticky und that the first layer after the support looks better?** 
I just switched from Simplyfy3D to PS and in S3D I had never such problems using support as in PS. And btw .. There where less option to configure the support. 

So hopefully someone can give some advice how to get better results with support und maybe there are some enhancements in upcoming releases of PS ? 

regards
   Dominik

JohnnyDeer, 2020-04-27:
Hi, could you please attach 3mf file of example you show on photo?

moelski, 2020-04-28:
> 3mf file of example 
Sure. Here it is. 

[Beer_Cap_Gun_left_half_32mm_Steel_Axle_v1.zip](https://github.com/prusa3d/PrusaSlicer/files/4543062/Beer_Cap_Gun_left_half_32mm_Steel_Axle_v1.zip)

That´s the way I tried to print it. But as I mentioned before using Layer Sync won´t make it better because the the first layer after the support sticks.

bubnikv, 2021-03-18:
To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio. 

[image]

Implemented with ceea9de8b8b1d4b35bdd1180051033680c405ba1 f01f02154c617cc24835475da419ed6322f2ceec

closed 2021-03-18 by bubnikv