prusa3d/PrusaSlicer #4128: conical support / angled support structures

IMightGitIt, 2020-04-26:
### Version
2.2.0+win64

### Operating system type + version
Windows 64 bit

### 3D printer brand / version + firmware version (if known)
Creality Cr10S, Marlin 2.0.3

### Feature Request Description
Enable a feature similar to the Conical Support in Cura.
In other words, a feature which creates support structures at an angle, to allow smaller supports at the print bed to expand up to a much bigger area when touching the model. Not to be confused with Support Trees which almost seems to do the opposite.
[image]

This feature saves both printing time and filament, and will in some cases make it much easier to get a grab on the support for easier removal.

_Is this a new feature request?_
Yes

rangoDJ, 2020-10-11:
agreed. this feature is not part of the latest cura 4.7 as an experimental feature. it would be very beneficial to include such into the superslicer.

stefanct, 2020-10-19:
I was looking for that as well; and [this guy too](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/angled-supports-not-from-build-plate-possible/). I am stealing his picture because it more precisely shows that I have in mind.

[image]

Use case (my first print on my own prusa mini): On the inside of a pipe I have two opposite overhangs thus I cannot rotate it to remove the need for support. Also, removing support that goes through the whole pipe would be really awkward if not impossible.

In that thread it was also mentioned, that prusa-slicer supports that for the SL1. Since SLA requires completely different supports, the code obviously can't be reused as is, but I made a screenshot anyway in case somebody is curious:

[image]

bttguy, 2021-02-01:
+1, conical support is really good in cura, would like to see PS is catching up in support settings/generating algorithm

DeflateAwning, 2022-11-27:
+1, conical support is an absolute must! The only reason I could think this wouldn't be implemented is if the company implementing it is the same company that earns from every roll of filament they sell :)

DeflateAwning, 2026-09-02:
Seems reasonable. With organic supports the last few years, I think this issue is basically unnecessary now. Can be closed.