prusa3d/PrusaSlicer #4130: Wrong bridging speed for small perimeters

bstoltefuss, 2020-04-27:
### Version
2.2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega-S

### Behavior
Small bridges (like those used in this temp tower: https://www.thingiverse.com/thing:2615842) will correctly trigger bridge detection, meaning the structure is correctly detected as "bridge infill" and "overhang perimeter". But when using 3 shells the innermost shell is printed with "small perimeter" speed instead of "bridge speed". The inner shell probably falls into the category "small perimeter" but i think "bridge speed" shoul have precedence. Otherwise correct bridge speed for all shells/infills can only be assured by setting "small perimeter" speed to "bridge speed" (which would make "small perimeter speed" useless.

[image]

[image]

#### Project File (.3MF) where problem occurs
[tempTower190to220.zip](https://github.com/prusa3d/PrusaSlicer/files/4539049/tempTower190to220.zip)

JohnnyDeer, 2020-04-27:
Hi, issue reproduced and confirmed, thank you for reporting.

chilman408, 2020-11-28:
Hello, having this issue with prints... how do we overcome this bug?

jschuh, 2021-02-14:
I have a trivial PR (#6044) to fix this, assuming that my approach is deemed the correct solution.

closed 2026-09-29 by prusaslicer-bot[bot]