prusa3d/PrusaSlicer #4135: Gtk3 issue on 4k+ screens

xarbit, 2020-04-27:
### Version
2.2.0

### Operating system type + version
Fedora 31 and Fedora 32, others also had this on Linux Mint.

### Behavior
If PrusaSlicer was built using wxGtk3 and you have a high resolution screen (4K/UHD), the screen with the platter does not scale with the rest of the application.  
See screenshot. 

[image]

This issue does not appear when built with wxGtk2. I assume it is a bug with wxWidgets

### How to reproduce.
1. use 4k or larger screen. 
2. go to display settings under linux and set scale to 200%
[image]
3. Build PrusaSlicer with wxGtk3
4. start PrusaSlicer.

xarbit, 2020-04-28:
I just noticed that there already is a similar issue open.
https://github.com/prusa3d/PrusaSlicer/issues/3245

Sorry for the duplication, I did not see it before.  

For what it is worth, I created an issue on the Fedora/RedHat issue tracker and switched to using wxGtk2 on the flatpak package on Flathub.  

I do hope these wxGtk3 issues solved so we don't have to continue to utilize the old & outdated Gtk2 for ages. I mean, Gtk4 is just around the corner.

Kolous, 2020-10-25:
Same issue here. Ubuntu 20.10, 4k display scale 200%.

HiFiJ, 2020-11-30:
Hi everyone, 

ongoing issue with version PrusaSlicer-2.3.0-beta1+linux-x64-GTK3-202011281134.AppImage on OpenSuse Tumbleweed, rendering the application non-functioning. Using the GTK2 version as a workaround so far still works.
Thanks for your efforts on PrusaSlicer!

bubnikv, 2020-12-15:
Fixed with a13b35ce92efed202ef43f02debb83dfb533d9b0

closed 2020-12-15 by bubnikv

lukasmatena, 2020-12-18:
This should be fixed in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Let us know if it isn't. Thanks.

nordurljosahvida, 2020-12-18:
Fixed for me in GTK3 appimage