prusa3d/PrusaSlicer #4137: Strange artifacts on perimeter when slicing

jsikorsky, 2020-04-28:
### Version
Version: 2.2.0+win64
Build: PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 17,070MB

### 3D printer brand / version + firmware version (if known)
Creality Ender 3, but not related to printer IMHO

### Behavior
*  When this particular model is sliced, I see very strange pieces of external perimeter on some layers. These pieces just stick into the air, and on some layers they are event not connected to the model at all
* I played with some settings (i.e. retraction, layer height), it's completely possible it's some kind of misconfiguration, but I was not able to figure it out
* I inspected the model and see nothing strange in these parts, Prusa Slicer does not report any problems, also Cura slicer make no such artifacts
* I didn't encounter such behaviour in any other model

#### Project File (.3MF) where problem occurs

[kazooBody.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4546107/kazooBody.3mf.zip)
[kazooBody.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/4546108/kazooBody.gcode.zip)
[config.zip](https://github.com/prusa3d/PrusaSlicer/files/4546092/config.zip)

[image]

[image]

rtyr, 2020-04-28:
Disable "Detect thin walls" parameter, it will fix it.

Duplicate of #4110, #340, #149, #3611. #227, #2583. Closing.

closed 2020-04-28 by rtyr

jsikorsky, 2020-04-28:
thanks a lot, really appreciate it!