prusa3d/PrusaSlicer #4138: app should attach filament settings to a given printer

tedder, 2020-04-28:
### Version
2.2.0+linux-x64

### Operating system type + version
Ubuntu 19.10. Gnome. Doesn't matter for this.

### 3D printer brand / version + firmware version (if known)
1. Prusa MK3S MMU2S
2. Prusa MK3S

### Behavior
This should be a fairly easy scenario to understand, giving a printing setup with two printers like this.

When I switch from, say, the MMU printer to the non-MMU printer, the filament gets reset to "prusament PLA". If I switch back to the MMU printer, all five extruders are set to "prusament PLA".

I can change the colors and filament types, set up my job, and then swap printers. When I come back to the MMU printer, all the extruder settings are wiped out again.

Much of this is saved in a 3mf file, but a printer's chosen filaments tend to remain the same, and it's unfortunate to lose those settings every time.

One way to do this would be to include the filament type in the "printer settings" -> "extruder N" area, where the extruder color currently resides.

#### Project File (.3MF) where problem occurs
n/a, but here are some screenshots.

starting out on the MMU2 printer:
[image]

changing some filament types:
[image]

switching to my non-MMU2 printer:
[image]

switching back to the MMU2 printer, all filament settings are gone:
[image]

tedder, 2020-04-28:
Seems related to #2653, fifth item in #2591.

mik-at, 2020-09-04:
+1
pls fix

mik-at, 2026-09-10:
This issue is still active in PrusaSlicer 2.9.6. Would really love if this gets finally fixed.