prusa3d/PrusaSlicer #4139: Saving print or filament preset changes to project is unsustainable over multiple changes

gfowley, 2020-04-28:
### Version
2.2.0+linux-x64

### Operating system type + version
Linux
Ubuntu 18.04 LTS

### 3D printer brand / version + firmware version (if known)
Prusa MK2S

### Behavior

When a project is open, attempts to save changes to print or filament settings results in error:
**Cannot overwrite an external profile.**

#### To reproduce

Open a project file (.3mf) that has custom print or filament presets. Change preset settings, try to save changes to settings from settings tab.

#### Expectation

Settings be saved without rename to project (only), preferably without repeated parens nesting.

#### Actual Result

User does must do one of:

- Rename setting and save, creating a new setting. This will be saved both in global user presets. Save project. Quit. Opening project again results in user confusion about presets. If preset name kept parens, name confusion is worse. Repeated changes require new names and quickly increase preset name pollution. 
- Do no rename or save setting. Save project, quit, ignore warning about unsaved setting. Open project, preset name is now nested in multiple parens. Repeated changes increase parens nesting quickly making preset name unreadable. 

Neither option available to user is sustainable.

gfowley, 2020-04-28:
They may be some overlap/interaction with issue #3238

rtyr, 2021-09-05:
What do you think about current state? Can we close this issue?

You can try https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1.