prusa3d/PrusaSlicer #4141: Fail Check, and if ignore it and install, segfault when slice any object

sl1pkn07, 2020-04-28:
### Version

version_2.2.0.191.gdb49a4516

### Operating system type + version
Archlinux

### 3D printer brand / version + firmware version (if known)
not aplicable

### Behavior
* _Describe the problem_
make check fail (ctest -V)

~~~
test 4
    Start 4: fff_print_tests

4: Test command: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/fff_print/fff_print_tests
4: Test timeout computed to be: 10000000
4: Testing Scenario: init_print functionality
4:     Given: A default config
4:      When: init_print is called with a single mesh.
4:      Then: One mesh/printobject is in the resulting Print object.
4:     Given: A default config
4:      When: init_print is called with a single mesh.
4:      Then: print.process() doesn't crash.
4: fff_print_tests: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/src/libslic3r/GCode/ToolOrdering.cpp:403: void Slic3r::ToolOrdering::fill_wipe_tower_partitions(const Slic3r::PrintConfig&, coordf_t, coordf_t): Assertion `! m_layer_tools.empty() && m_layer_tools.front().has_wipe_tower' failed.
4: 
4: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
4: fff_print_tests is a Catch v2.9.2 host application.
4: Run with -? for options
4: 
4: -------------------------------------------------------------------------------
4: Scenario: init_print functionality
4:       Given: A default config
4:        When: init_print is called with a single mesh.
4:        Then: print.process() doesn't crash.
4: -------------------------------------------------------------------------------
4: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/tests/fff_print/test_data.cpp:310
4: ...............................................................................
4: 
4: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/tests/fff_print/test_data.cpp:310: FAILED:
4: due to a fatal error condition:
4:   SIGABRT - Abort (abnormal termination) signal
4: 
4: ===============================================================================
4: test cases: 1 | 1 failed
4: assertions: 2 | 1 passed | 1 failed
4: 
4/5 Test #4: fff_print_tests ..................Child aborted***Exception:   0.67 sec
~~~
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_

~~~
  cmake ../PrusaSlicer \
      -DCMAKE_BUILD_TYPE=none \
      -DCMAKE_INSTALL_PREFIX=/usr \
      -DCMAKE_INSTALL_LIBDIR=lib \
      -DSLIC3R_FHS=ON \
      -DSLIC3R_PCH=OFF \
      -DSLIC3R_WX_STABLE=ON \
      -DSLIC3R_GTK=3 \
      -DSLIC3R_STATIC=OFF \
      -DOpenGL_GL_PREFERENCE=GLVND
~~~
* _Expected Results_

Check OK
* _Actual Results_
segfault

if ignore the test, and install it. when slice any objet (any, no matter what)
Prusa slicer gets Segfault

~~~
prusa-slicer: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/src/libslic3r/GCode/ToolOrdering.cpp:403: void Slic3r::ToolOrdering::fill_wipe_tower_partitions(const Slic3r::PrintConfig&, coordf_t, coordf_t): La declaración `! m_layer_tools.empty() && m_layer_tools.front().has_wipe_tower' no se cumple.
Abortado (`core' generado)
~~~

wxwidgets 3.0.4 over GTK3 3.24.18
cmake 3.17.1
gcc 9.3.0

sl1pkn07, 2020-05-30:
same with 2.3.0.alpha0.134

also test 5 fail

~~~
test 5
    Start 5: sla_print_tests

5: Test command: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/sla_print/sla_print_tests
5: Test timeout computed to be: 10000000
5: Testing Pillar pairhash should be unique
5: Passed in 0.003116 [seconds]
5: 
5: Testing Support point generator should be deterministic if seeded
5: Passed in 0.197815 [seconds]
5: 
5: Testing Flat pad geometry is valid
5: Passed in 0.004813 [seconds]
5: 
5: Testing WingedPadGeometryIsValid
5: Passed in 0.006941 [seconds]
5: 
5: Testing FlatPadAroundObjectIsValid
5: Passed in 0.079693 [seconds]
5: 
5: Testing WingedPadAroundObjectIsValid
5: Passed in 0.08585 [seconds]
5: 
5: Testing ElevatedSupportGeometryIsValid
5: Passed in 1.56501 [seconds]
5: 
5: Testing FloorSupportGeometryIsValid
5: Passed in 1.02906 [seconds]
5: 
5: Testing ElevatedSupportsDoNotPierceModel
5: 
5: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
5: sla_print_tests is a Catch v2.9.2 host application.
5: Run with -? for options
5: 
5: -------------------------------------------------------------------------------
5: ElevatedSupportsDoNotPierceModel
5: -------------------------------------------------------------------------------
5: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_print_tests.cpp:140
5: ...............................................................................
5: 
5: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_test_utils.cpp:46: FAILED:
5:   REQUIRE( notouch )
5: with expansion:
5:   false
5: 
5: Testing FloorSupportsDoNotPierceModel
5: Passed in 3.83622 [seconds]
5: 
5: Testing InitializedRasterShouldBeNONEmpty
5: Passed in 0.000882 [seconds]
5: 
5: Testing MirroringShouldBeCorrect
5: Passed in 0.003873 [seconds]
5: 
5: Testing RasterizedPolygonAreaShouldMatch
5: Passed in 0.091671 [seconds]
5: 
5: Testing Triangle mesh conversions should be correct
5: Passed in 3.2e-05 [seconds]
5: 
5: Testing Raycaster - find intersections of a line and cylinder
5: Passed in 1.6e-05 [seconds]
5: 
5: Testing Raycaster with loaded drillholes
5: Passed in 4.58136 [seconds]
5: 
5: ===============================================================================
5: test cases:    16 |    15 passed | 1 failed
5: assertions: 10824 | 10823 passed | 1 failed
5: 
5/5 Test #5: sla_print_tests ..................***Failed   20.20 sec
~~~

sl1pkn07, 2020-06-20:
any notice of this?

greetings

sl1pkn07, 2020-06-20:
https://github.com/prusa3d/PrusaSlicer/commit/bb62f36df3915f125c7460c6c12ab88a47efe798

https://github.com/prusa3d/PrusaSlicer/commit/bb62f36df3915f125c7460c6c12ab88a47efe798#diff-498aa9d7dcf8df8995c4ed3f9b9ba5a1L364

https://github.com/prusa3d/PrusaSlicer/commit/bb62f36df3915f125c7460c6c12ab88a47efe798#diff-d54379a7db2c0147be63c78752889bd3R48

why is the `if (!notouch)` commented?

sl1pkn07, 2020-06-20:
Works with clang (10.0.0) but explode with GCC (10.1.0)

also this STL explode with

~~~
prusa-slicer: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/src/admesh/shared.cpp:217: bool stl_validate(const stl_file *, const indexed_triangle_set &): La declaración `its.vertices.empty() || its.indices.size() == stl->stats.number_of_facets' no se cumple.
Abortado (`core' generado)
~~~
(sorry, not works with LC_ALL=C)

[dial_indicator_holder-mod.zip](https://github.com/prusa3d/PrusaSlicer/files/4808592/dial_indicator_holder-mod.zip)
(please rename to .stl)

greertings

StefanHamminga, 2020-08-29:
Same here on Arch Linux, test 5 fails:

```
5: Testing ElevatedSupportsDoNotPierceModel
5: 
5: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
5: sla_print_tests is a Catch v2.9.2 host application.
5: Run with -? for options
5: 
5: -------------------------------------------------------------------------------
5: ElevatedSupportsDoNotPierceModel
5: -------------------------------------------------------------------------------
5: ../tests/sla_print/sla_print_tests.cpp:142
5: ...............................................................................
5: 
5: ../tests/sla_print/sla_test_utils.cpp:46: FAILED:
5:   REQUIRE( notouch )
5: with expansion:
5:   false
```

System CFLAGS (seem to be ignored):

``` shell
CFLAGS=-march=haswell -mtune=skylake -O3 --param=ssp-buffer-size=32 -fdevirtualize-speculatively -fno-signed-zeros -fno-trapping-math -fassociative-math  -pipe -fno-plt -fdiagnostics-color=always -fstack-protector-strong -D_FORTIFY_SOURCE=2
CXXFLAGS=-march=haswell -mtune=skylake -O3 --param=ssp-buffer-size=32 -fdevirtualize-speculatively -fno-signed-zeros -fno-trapping-math -fassociative-math  -pipe -fno-plt -fdiagnostics-color=always -fstack-protector-strong -D_FORTIFY_SOURCE=2
CPPFLAGS=-fstack-protector-strong -D_FORTIFY_SOURCE=2
LDFLAGS=-fstack-protector-strong -D_FORTIFY_SOURCE=2 -march=haswell -mtune=skylake -O3 --param=ssp-buffer-size=32 -fdevirtualize-speculatively -fno-signed-zeros -fno-trapping-math -fassociative-math  -Wl,--sort-common,--as-needed,-z,relro -Wl,--check-sections -Wl,--gc-sections

```

CMake output for the build:

```
==> Updated version: prusa-slicer-git 2.3.0.alpha0.r336.g143e3a6a3-1
==> Starting build()...
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: ON
-- SLIC3R_GTK: 3
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: OFF
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_WX_STABLE: ON
-- SLIC3R_STATIC: OFF
-- CMAKE_PREFIX_PATH: (default)
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
[... 55 lines cut ...]
-- OpenVDB ABI Version: 7
-- Found Boost: /usr/lib64/cmake/Boost-1.72.0/BoostConfig.cmake (found version "1.72.0") found components: iostreams system 
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Performing Test LLVM_HAS_ATOMICS
-- Performing Test LLVM_HAS_ATOMICS - Success
-- OpenVDB libraries: /usr/lib/libopenvdb.so
-- Miniz NOT found in system, using bundled version...
-- Using qhull from system.
-- IGL NOT found, using bundled version...
-- __cplusplus is 201402
--   --> Do not link with Boost.Thread
-- Found Boost: /usr/lib64/cmake/Boost-1.72.0/BoostConfig.cmake (found suitable version "1.72.0", minimum required is "1.48")  
-- Boost include dirs: /usr/include
-- Boost libraries:    
-- Found wxWidgets: -pthread;;;-lwx_baseu-3.0;-lwx_gtk3u_core-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_gl-3.0 (found suitable version "3.0.5", minimum required is "3.0") 
-- wx-config path: /usr/bin/wx-config
-- wx libs: -pthread;;;-lwx_baseu-3.0;-lwx_gtk3u_core-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_gl-3.0;/usr/lib/libpng.so;/usr/lib/libz.so;/usr/lib/libexpat.so;/usr/lib/librt.so
-- Configuring done
-- Generating done
-- Build files have been written to: /storage/tmp/makepkg/build/prusa-slicer-git/src/PrusaSlicer/build
```

The build warnings generated:
[warnings.txt](https://github.com/prusa3d/PrusaSlicer/files/5144841/warnings.txt)

StefanHamminga, 2020-08-29:
Update: For me the above mentioned work-around of switching to Clang doesn't work, the result is exactly the same.

CMake output:

```
-- The C compiler identification is Clang 10.0.1
-- The CXX compiler identification is Clang 10.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: ON
-- SLIC3R_GTK: 3
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: OFF
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_WX_STABLE: ON
-- SLIC3R_STATIC: OFF
-- CMAKE_PREFIX_PATH: (default)
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
CMake Warning (dev) at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (DBUS) does
  not match the name of the calling package (DBus).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindDBus.cmake:59 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:169 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found DBUS: /usr/include/dbus-1.0;/usr/lib/dbus-1.0/include  
-- Found Boost: /usr/lib64/cmake/Boost-1.72.0/BoostConfig.cmake (found suitable version "1.72.0", minimum required is "1.64.0") found components: system filesystem thread log locale regex chrono atomic date_time 
-- Boost::boost exists
-- Found TBB: /usr/include (found version "2020.3")  
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /usr/include
--   TBB_DEFINITIONS         = 
--   TBB_LIBRARIES           = /usr/lib/libtbb.so
--   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = 
--   TBB_DEFINITIONS_RELEASE = 
--   TBB_LIBRARIES_RELEASE   = /usr/lib/libtbb.so
-- Found CURL: /usr/lib/libcurl.so (found version "7.72.0") 
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.11") 
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.3") 
-- Found EXPAT: /usr/lib/libexpat.so (found version "2.2.9") 
-- Found PNG: /usr/lib/libpng.so (found version "1.6.37") 
CMake Warning (dev) at /usr/share/cmake-3.18/Modules/FindOpenGL.cmake:305 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMakeLists.txt:396 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: /usr/lib/libOpenGL.so   
-- Found GLEW: /usr/include (found version "2.2.0") 
-- Found NLopt in '/usr/lib'.
-- Using NLopt include directory '/usr/include'.
-- Using NLopt library '/usr/lib/libnlopt.so'.
-- Found OpenVDB: /usr/include (found suitable version "7.1.0", minimum required is "5.0") found components: openvdb 
-- OpenVDB ABI Version: 7
-- Found Boost: /usr/lib64/cmake/Boost-1.72.0/BoostConfig.cmake (found version "1.72.0") found components: iostreams system 
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Performing Test LLVM_HAS_ATOMICS
-- Performing Test LLVM_HAS_ATOMICS - Success
-- OpenVDB libraries: /usr/lib/libopenvdb.so
-- Miniz NOT found in system, using bundled version...
-- Using qhull from system.
-- IGL NOT found, using bundled version...
-- __cplusplus is 201402
--   --> Do not link with Boost.Thread
-- Found Boost: /usr/lib64/cmake/Boost-1.72.0/BoostConfig.cmake (found suitable version "1.72.0", minimum required is "1.48")  
-- Boost include dirs: /usr/include
-- Boost libraries:    
-- Found wxWidgets: -pthread;;;-lwx_baseu-3.0;-lwx_gtk3u_core-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_gl-3.0 (found suitable version "3.0.5", minimum required is "3.0") 
-- wx-config path: /usr/bin/wx-config
-- wx libs: -pthread;;;-lwx_baseu-3.0;-lwx_gtk3u_core-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_gl-3.0;/usr/lib/libpng.so;/usr/lib/libz.so;/usr/lib/libexpat.so;/usr/lib/librt.so
-- Configuring done
-- Generating done
-- Build files have been written to: /storage/tmp/makepkg/build/prusa-slicer-git/src/PrusaSlicer/build
```

sl1pkn07, 2020-08-29:
little hack for DBus check

~~~
diff --git a/cmake/modules/FindDBus.cmake b/cmake/modules/FindDBus.cmake
index 1d0f29dd7..d54d4e516 100644
--- a/cmake/modules/FindDBus.cmake
+++ b/cmake/modules/FindDBus.cmake
@@ -56,4 +56,4 @@ FIND_PATH(DBUS_ARCH_INCLUDE_DIR
 SET(DBUS_INCLUDE_DIRS ${DBUS_INCLUDE_DIR} ${DBUS_ARCH_INCLUDE_DIR})
 
 INCLUDE(FindPackageHandleStandardArgs)
-FIND_PACKAGE_HANDLE_STANDARD_ARGS(DBUS REQUIRED_VARS DBUS_INCLUDE_DIRS DBUS_LIBRARIES)
\ No newline at end of file
+FIND_PACKAGE_HANDLE_STANDARD_ARGS(DBus REQUIRED_VARS DBUS_INCLUDE_DIRS DBUS_LIBRARIES)
\ No newline at end of file
~~~

~~~
  cd build
  cmake ../PrusaSlicer \
      -DCMAKE_BUILD_TYPE=none \
      -DCMAKE_INSTALL_PREFIX=/usr \
      -DCMAKE_INSTALL_LIBDIR=lib \
      -DSLIC3R_FHS=ON \
      -DSLIC3R_PCH=OFF \
      -DSLIC3R_WX_STABLE=ON \
      -DSLIC3R_GTK=3 \
      -DSLIC3R_STATIC=OFF \
      -DOpenGL_GL_PREFERENCE=GLVND \
      -DBUILD_TESTING=ON \
      -DCMAKE_CXX_COMPILER=clang++ \
      -DCMAKE_C_COMPILER=clang
~~~

~~~
-- The C compiler identification is Clang 10.0.1
-- The CXX compiler identification is Clang 10.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: ON
-- SLIC3R_GTK: 3
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: OFF
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_WX_STABLE: ON
-- SLIC3R_STATIC: OFF
-- CMAKE_PREFIX_PATH: (default)
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
[... 107 lines cut ...]
      -DSLIC3R_STATIC=OFF \
      -DOpenGL_GL_PREFERENCE=GLVND \
      -DBUILD_TESTING=ON \
      -DCMAKE_CXX_COMPILER=clang++ \
      -DCMAKE_C_COMPILER=clang

  make
}

check() {
  cd build
  ctest -V
}

package () {
    make -C build DESTDIR="${pkgdir}" install

    # Desktop file
    install -Dm644 "${srcdir}/prusa-slicer.desktop" "${pkgdir}/usr/share/applications/prusa-slicer.desktop"

    install -Dm644 "${pkgdir}/usr/share/PrusaSlicer/icons/PrusaSlicer_192px.png" "${pkgdir}/usr/share/pixmaps/prusa-slicer.png"
}
~~~

</details>

StefanHamminga, 2020-08-29:
I just did another build, replacing all CXXFLAGS with just "-O2" and it passes the tests now.

@sl1pkn07 Could you post your CXXFLAGS (and similar)?

sl1pkn07, 2020-08-29:
~~~
#-- Compiler and Linker Flags
CPPFLAGS="-D_FORTIFY_SOURCE=2"
#CFLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fno-plt"
CFLAGS="-march=native -O2 -pipe -fno-plt"
#CXXFLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fno-plt"
CXXFLAGS="-march=native -O2 -pipe -fno-plt"
LDFLAGS="-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"
#-- Make Flags: change this for DistCC/SMP systems
#MAKEFLAGS="-j2"
MAKEFLAGS="-j40"
#-- Debugging flags
DEBUG_CFLAGS="-g -fvar-tracking-assignments"
DEBUG_CXXFLAGS="-g -fvar-tracking-assignments"
~~~

StefanHamminga, 2020-08-29:
And that works now? Or did it fail the test with those? In case it failed it seems the overlap is in extended instruction use (native / haswell) and disabling the [PLT](https://gcc.gnu.org/onlinedocs/gcc/Code-Gen-Options.html).

sl1pkn07, 2020-08-29:
works. but only if is builded with clang

StefanHamminga, 2020-08-29:
Can you check with gcc? Unsetting LDFLAGS and CPPFLAGS, setting CFLAGS and CXXFLAGS to "-O2"? That worked for me.

sl1pkn07, 2020-08-29:
nope. fails with gcc (10.20) for me

StefanHamminga, 2020-09-01:
Turns out, last time it built and tested ok, but segfaults on running:

```
0x00005555562ec0e3 in ?? ()
#0  0x00005555562ec0e3 in ?? ()
#1  0x00005555562c1d03 in ?? ()
#2  0x0000555555eeb313 in ?? ()
#3  0x0000555555eebc35 in ?? ()
#4  0x0000555555eee141 in ?? ()
#5  0x0000555555f1864d in ?? ()
#6  0x0000555555cc2ae7 in ?? ()
#7  0x0000555555cc3842 in ?? ()
#8  0x00007ffff6dbb4c1 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () from /usr/lib/libwx_baseu-3.0.so.0
#9  0x00007ffff6dbb84b in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () from /usr/lib/libwx_baseu-3.0.so.0
#10 0x00007ffff6dbb8e1 in wxEvtHandler::TryHereOnly(wxEvent&) () from /usr/lib/libwx_baseu-3.0.so.0
#11 0x00007ffff6dbb994 in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /usr/lib/libwx_baseu-3.0.so.0
#12 0x00007ffff6dbba32 in wxEvtHandler::ProcessEvent(wxEvent&) () from /usr/lib/libwx_baseu-3.0.so.0
#13 0x00007ffff6c33cf9 in wxAppConsoleBase::ProcessIdle() () from /usr/lib/libwx_baseu-3.0.so.0
#14 0x00007ffff6889ad5 in wxAppBase::ProcessIdle() () from /usr/lib/libwx_gtk3u_core-3.0.so.0
#15 0x00007ffff67b7306 in wxApp::DoIdle() () from /usr/lib/libwx_gtk3u_core-3.0.so.0
#16 0x00007ffff67b7424 in ?? () from /usr/lib/libwx_gtk3u_core-3.0.so.0
#17 0x00007ffff2853340 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#18 0x00007ffff28a11d9 in ?? () from /usr/lib/libglib-2.0.so.0
#19 0x00007ffff2851c03 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#20 0x00007ffff2d754af in gtk_dialog_run () from /usr/lib/libgtk-3.so.0
#21 0x00007ffff685d21e in wxMessageDialog::ShowModal() () from /usr/lib/libwx_gtk3u_core-3.0.so.0
#22 0x0000555555cb9047 in ?? ()
#23 0x0000555555cb91ee in ?? ()
#24 0x0000555555cc1555 in ?? ()
#25 0x0000555555cc1ec6 in ?? ()
#26 0x00007ffff6cc5822 in wxEntry(int&, wchar_t**) () from /usr/lib/libwx_baseu-3.0.so.0
#27 0x0000555555764f36 in ?? ()
#28 0x00005555557501f5 in ?? ()
#29 0x00007ffff5ec5152 in __libc_start_main () from /usr/lib/libc.so.6
#30 0x000055555575de0e in ?? ()
```

rtyr, 2021-02-16:
Is this issue still valid? Can we close it?

sl1pkn07, 2021-02-16:
let me check

sl1pkn07, 2021-02-16:
failed in 3 checks

~~~
test 2
    Start 2: libslic3r_tests

2: Test command: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/libslic3r/libslic3r_tests
2: Test timeout computed to be: 10000000
2: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/libslic3r/libslic3r_tests: symbol lookup error: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/libslic3r/libslic3r_tests: undefined symbol: _ZN3tbb10interface78internal15task_arena_base24internal_max_concurrencyEPKNS0_10task_arenaE
2/5 Test #2: libslic3r_tests ..................***Failed    0.02 sec
test 3
    Start 3: slic3rutils_tests

3: Test command: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/slic3rutils/slic3rutils_tests " exclude:[NotWorking]"
3: Test timeout computed to be: 10000000
3: Filters: exclude:[NotWorking]
3: No test cases matched 'exclude:[NotWorking]'
3: ===============================================================================
3: No tests ran
3: 
3/5 Test #3: slic3rutils_tests ................   Passed    0.03 sec
test 4
    Start 4: fff_print_tests

4: Test command: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/fff_print/fff_print_tests
4: Test timeout computed to be: 10000000
4: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/fff_print/fff_print_tests: symbol lookup error: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/fff_print/fff_print_tests: undefined symbol: _ZN3tbb10interface78internal15task_arena_base24internal_max_concurrencyEPKNS0_10task_arenaE
4/5 Test #4: fff_print_tests ..................***Failed    0.01 sec
test 5
    Start 5: sla_print_tests

5: Test command: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/sla_print/sla_print_tests
5: Test timeout computed to be: 10000000
5: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/sla_print/sla_print_tests: symbol lookup error: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/sla_print/sla_print_tests: undefined symbol: _ZN3tbb10interface78internal15task_arena_base24internal_max_concurrencyEPKNS0_10task_arenaE
5/5 Test #5: sla_print_tests ..................***Failed    0.01 sec

40% tests passed, 3 tests failed out of 5

Total Test time (real) =   0.27 sec

The following tests FAILED:
          2 - libslic3r_tests (Failed)
          4 - fff_print_tests (Failed)
          5 - sla_print_tests (Failed)
Errors while running CTest
~~~

i need more test

sl1pkn07, 2021-03-22:
hi, i found the poblem of my latest report.

test is linked with wrong version of TBB library, because not preserve/use the library path founded by cmake

[image]
cmake fund the rigth library

but the ctest is linked whith wrong one

~~~
└───╼  ldd /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/sla_print/sla_print_tests
foo
        libtbb.so.2 => /opt/intel/opencl-1.2-4.5.0.8/lib64/libtbb.so.2 (0x00007f67f547f000)
foo
 ~~~

(the library is in the LD_LIBRARY_PATH because i added it with my `ld.so.conf.d` file)

the ctest should be linked with the library funded with cmake

tested again (commit 1c2d26457) with

~~~
LD_LIBRARY_PATH=/usr/lib ctest -V
~~~

and now fail with:

~~~
5: Test command: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/sla_print/sla_print_tests
5: Test timeout computed to be: 10000000
5: Testing Pillar pairhash should be unique
5: Passed in 0.003393 [seconds]
5: 
5: Testing Support point generator should be deterministic if seeded
5: Passed in 0.183629 [seconds]
5: 
5: Testing Flat pad geometry is valid
5: Passed in 0.004474 [seconds]
5: 
5: Testing WingedPadGeometryIsValid
5: Passed in 0.006989 [seconds]
5: 
5: Testing FlatPadAroundObjectIsValid
5: Passed in 0.074247 [seconds]
5: 
5: Testing WingedPadAroundObjectIsValid
5: Passed in 0.084494 [seconds]
5: 
5: Testing ElevatedSupportGeometryIsValid
5: Passed in 6.32604 [seconds]
5: 
5: Testing FloorSupportGeometryIsValid
5: Passed in 5.5236 [seconds]
5: 
5: Testing ElevatedSupportsDoNotPierceModel
5: [2021-03-22 19:34:58.735011] [0x00007f4216bfa640] [error]   Pinhead width is negative!
5: Passed in 25.0894 [seconds]
5: 
5: Testing FloorSupportsDoNotPierceModel
5: [2021-03-22 19:35:16.742796] [0x00007f42147f1640] [error]   Pinhead width is negative!
5: Passed in 15.163 [seconds]
5: 
5: Testing InitializedRasterShouldBeNONEmpty
5: Passed in 0.000898 [seconds]
5: 
5: Testing MirroringShouldBeCorrect
5: Passed in 0.003693 [seconds]
5: 
5: Testing RasterizedPolygonAreaShouldMatch
5: Passed in 0.091751 [seconds]
5: 
5: Testing Triangle mesh conversions should be correct
5: Passed in 2.8e-05 [seconds]
5: 
5: Testing halfcone test
5: Passed in 0.000706 [seconds]
5: 
5: Testing Test concurrency
5: Passed in 5.2e-05 [seconds]
5: 
5: Testing Overhanging point should be supported
5: Passed in 0.002199 [seconds]
5: 
5: Testing Overhanging horizontal surface should be supported
5: Passed in 0.000801 [seconds]
5: 
5: Testing Overhanging edge should be supported
5: Passed in 0.001881 [seconds]
5: 
5: Testing Hollowed cube should be supported from the inside
5: Passed in 1.64427 [seconds]
5: 
5: Testing Two parallel plates should be supported
5: Passed in 0.006415 [seconds]
5: 
5: Testing Raycaster - find intersections of a line and cylinder
5: Passed in 2.2e-05 [seconds]
5: 
5: ===============================================================================
5: All tests passed (13831 assertions in 22 test cases)
5: 
5/5 Test #5: sla_print_tests ..................   Passed   54.27 sec

60% tests passed, 2 tests failed out of 5

Total Test time (real) =  62.29 sec

The following tests FAILED:
          2 - libslic3r_tests (Subprocess aborted)
          3 - slic3rutils_tests (Not Run)
Errors while running CTest
~~~

greetings

sl1pkn07, 2021-04-24:
more test failed 9f4e9da5c

~~~
2: Test command: /tmp/makepkg/sl1-prusa-slicer-git/src/build/tests/libslic3r/libslic3r_tests
2: Test timeout computed to be: 10000000
2: Testing sort_remove_duplicates
2: Passed in 1e-05 [seconds]
2: 
2: Testing string_printf
2: Empty format with empty data should return empty string
2: String output length should be the same as input
2: String format should be interpreted as with sprintf
2: String format should survive large input data
2: Passed in 7.4e-05 [seconds]
2: 
2: Testing Scenario: Reading 3mf file
2:     Given: umlauts in the path of the file
2:      When: 3mf model is read
2:      Then: load should succeed
2: Passed in 0.007839 [seconds]
2: 
2: Testing Scenario: Export+Import geometry to/from 3mf file cycle
2:     Given: world vertices coordinates before save
2:      When: model is saved+loaded to/from 3mf file
2:      Then: world vertices coordinates after load match
2: Passed in 0.037776 [seconds]
2: 
2: Testing Building a tree over a box, ray caster and closest query
2: Passed in 0.000304 [seconds]
2: 
2: Testing Scenario: Constant offset
2:     Given: 20mm box
2:      When: Slic3r::offset()
2: plus 1mm, miter 2x
2:      Then: Area is 22^2mm2
2:     Given: 20mm box
2:      When: Slic3r::offset()
2: minus 1mm, miter 2x
2:      Then: Area is 18^2mm2
2:     Given: 20mm box
2:      When: Slic3r::offset()
2: plus 1mm, miter 1.5x
2:      Then: Area is 22^2mm2
2:     Given: 20mm box
2:      When: Slic3r::offset()
2: minus 1mm, miter 1.5x
2:      Then: Area is 18^2mm2
2:     Given: 20mm box
2:      When: Slic3r::offset()
2: plus 1mm, miter 1.2x
2:      Then: Area is 22^2mm2
2:     Given: 20mm box
2:      When: Slic3r::offset()
2: minus 1mm, miter 1.2x
2:      Then: Area is 18^2mm2
2:     Given: 20mm box
2:      When: Slic3r::variable_offset_outer/inner
2: plus 1mm, miter 2x
2:      Then: Area is 22^2mm2
2:     Given: 20mm box
2:      When: Slic3r::variable_offset_outer/inner
2: minus 1mm, miter 2x
2:      Then: Area is 18^2mm2
2:     Given: 20mm box
2:      When: Slic3r::variable_offset_outer/inner
2: plus 1mm, miter 1.5x
2:      Then: Area is 22^2mm2
2:     Given: 20mm box
2:      When: Slic3r::variable_offset_outer/inner
2: minus 1mm, miter 1.5x
[... 925 lines cut ...]
4: fff_print_tests: /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:281: void Eigen::PlainObjectBase<Derived>::resize(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<int, 3, 1, 2>; Eigen::Index = long int]: Assertion `(!(RowsAtCompileTime!=Dynamic) || (rows==RowsAtCompileTime)) && (!(ColsAtCompileTime!=Dynamic) || (cols==ColsAtCompileTime)) && (!(RowsAtCompileTime==Dynamic && MaxRowsAtCompileTime!=Dynamic) || (rows<=MaxRowsAtCompileTime)) && (!(ColsAtCompileTime==Dynamic && MaxColsAtCompileTime!=Dynamic) || (cols<=MaxColsAtCompileTime)) && rows>=0 && cols>=0 && "Invalid sizes when resizing a matrix or array."' failed.
4: 
4: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
4: fff_print_tests is a Catch v2.13.3 host application.
4: Run with -? for options
4: 
4: -------------------------------------------------------------------------------
4: Scenario: Original Slic3r Skirt/Brim tests
4:       Given: A default configuration
4:        When: Skirt area is smaller than the brim
4:        Then: Gcode generates
4: -------------------------------------------------------------------------------
4: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/tests/fff_print/test_skirt_brim.cpp:107
4: ...............................................................................
4: 
4: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/tests/fff_print/test_skirt_brim.cpp:107: FAILED:
4: due to a fatal error condition:
4:   SIGABRT - Abort (abnormal termination) signal
4: 
4: ===============================================================================
4: test cases:  19 |  18 passed | 1 failed
4: assertions: 350 | 349 passed | 1 failed
4: 
4/5 Test #4: fff_print_tests ..................Subprocess aborted***Exception:   4.46 sec
~~~

bubnikv, 2021-09-22:
We only support building on Linux using our static dependencies.
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md
If you do it differently, you are on your own. We accept pull requests fixing compilation issues.

closed 2021-09-22 by bubnikv

sl1pkn07, 2021-09-22:
None of the test failed is about static/dynamic dependencies or how is the projet builded

So closing the issue about you comment is invalid

If you read across the issue, you see one time works and others not. So you make regressions in the code and not fix it

Please reopen and fix the problems posted in the issue

Greetings

bubnikv, 2021-09-23:
We do not see your issue, thus we cannot fix it.
We will gladly accept a pull request.

sl1pkn07, 2021-09-23:
What part exactly you not see? Please copy@paste the code

Greetings

lukasmatena, 2021-09-23:
@sl1pkn07 We build statically against dependencies pulled by the script in deps folder. The tests don't fail, the application does not crash. You build against some unknown versions of those libraries, maybe mixing libraries built with gcc and clang, maybe with further incompatibilities in libc++/libstdc++ and who knows what else. That may lead to mysterious crashes and we don't have resources to untangle it and reason what exactly is wrong.

Of course there can be a different problem. But again, if we cannot reproduce it, we cannot fix it.

sl1pkn07, 2021-09-23:
as you wish, use the instructions on https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

~~~
Test project /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/build
    Start 1: libnest2d_tests
1/5 Test #1: libnest2d_tests ..................   Passed    0.22 sec
    Start 2: libslic3r_tests
2/5 Test #2: libslic3r_tests ..................***Failed    9.49 sec
[2021-09-23 19:35:13.974272] [0x00007f96f9329b80] [trace]   Initializing StaticPrintConfigs
Testing sort_remove_duplicates
Passed in 9e-06 [seconds]

Testing string_printf
Empty format with empty data should return empty string
String output length should be the same as input
String format should be interpreted as with sprintf
String format should survive large input data
Passed in 8.3e-05 [seconds]

Testing Scenario: Reading 3mf file
    Given: umlauts in the path of the file
     When: 3mf model is read
     Then: load should succeed
Passed in 0.001512 [seconds]

Testing Scenario: Export+Import geometry to/from 3mf file cycle
    Given: world vertices coordinates before save
     When: model is saved+loaded to/from 3mf file
     Then: world vertices coordinates after load match
Passed in 0.022034 [seconds]

Testing Scenario: 2D convex hull of sinking object
    Given: model
     When: model is rotated, scaled and set as sinking
     Then: 2D convex hull should match with reference

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libslic3r_tests is a Catch v2.13.6 host application.
Run with -? for options

-------------------------------------------------------------------------------
Scenario: 2D convex hull of sinking object
      Given: model
       When: model is rotated, scaled and set as sinking
       Then: 2D convex hull should match with reference
-------------------------------------------------------------------------------
/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/tests/libslic3r/test_3mf.cpp:117
...............................................................................

/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/tests/libslic3r/test_3mf.cpp:118: FAILED:
  REQUIRE( res )
with expansion:
  false

    Given: model
     When: model is rotated, scaled and set as sinking
Testing Building a tree over a box, ray caster and closest query
Passed in 3.3e-05 [seconds]

Testing Scenario: Constant offset
    Given: 20mm box
     When: Slic3r::offset()
plus 1mm, miter 2x
     Then: Area is 22^2mm2
    Given: 20mm box
     When: Slic3r::offset()
minus 1mm, miter 2x
     Then: Area is 18^2mm2
    Given: 20mm box
     When: Slic3r::offset()
[... 964 lines cut ...]
-- Miniz NOT found in system, using bundled version...
-- Using qhull from system.
-- [libigl] Looking for Eigen3
-- IGL found, using system version...
-- Using header-only CGAL
-- Targetting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Found GMP: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/libgmp.a  
-- Found MPFR: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/libmpfr.a  
-- Found Boost: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/cmake/Boost-1.75.0/BoostConfig.cmake (found suitable version "1.75.0", minimum required is "1.48")  
-- Boost include dirs: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/include/boost-1_75
-- Boost libraries:    
-- Using gcc version 4 or later. Adding -frounding-math
-- Found wxWidgets: -L/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib;-pthread;/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_gtk3u_gl-3.1.a;/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_gtk3u_html-3.1.a;/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_gtk3u_core-3.1.a;/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_baseu-3.1.a;libSM.so;libICE.so;libX11.so;libXext.so;gtk-3;gdk-3;z;pangocairo-1.0;pango-1.0;harfbuzz;atk-1.0;cairo-gobject;cairo;gdk_pixbuf-2.0;gio-2.0;gobject-2.0;glib-2.0;libOpenGL.so;libGLX.so;libGLU.so;libSM.so;libICE.so;libX11.so;libXext.so;gtk-3;gdk-3;z;pangocairo-1.0;pango-1.0;harfbuzz;atk-1.0;cairo-gobject;cairo;gdk_pixbuf-2.0;gio-2.0;gobject-2.0;glib-2.0;-lwx_gtk3u_core-3.1;libSM.so;libICE.so;libX11.so;libXext.so;gtk-3;gdk-3;z;pangocairo-1.0;pango-1.0;harfbuzz;atk-1.0;cairo-gobject;cairo;gdk_pixbuf-2.0;gio-2.0;gobject-2.0;glib-2.0;-lwx_baseu-3.1;libjpeg.a;libpng.a;libz.so;m;libtiff.a;libSM.so;libICE.so;libX11.so;libXext.so;gtk-3;gdk-3;z;pangocairo-1.0;pango-1.0;harfbuzz;atk-1.0;cairo-gobject;cairo;gdk_pixbuf-2.0;gio-2.0;gobject-2.0;glib-2.0;libz.so;-lwxregexu-3.1;secret-1;gio-2.0;gobject-2.0;glib-2.0;libc.so;-lpthread;dl (found suitable version "3.1.4", minimum required is "3.0") 
-- wx-config path: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/bin/wx-config
-- wx libs: -L/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib;-pthread;/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_gtk3u_gl-3.1.a;/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_gtk3u_html-3.1.a;/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_gtk3u_core-3.1.a;/tmp/makepkg/sl1-prusa-slicer-gi [... cut ...] lude/cairo;/usr/include/lzo;/usr/include/pixman-1;/usr/include/gdk-pixbuf-2.0;/tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/deps/build/destdir/usr/local/include;/usr/include/gio-unix-2.0;/usr/include/cloudproviders;/usr/include/atk-1.0;/usr/include/at-spi2-atk/2.0;/usr/include/dbus-1.0;/usr/lib/dbus-1.0/include;/usr/include/at-spi-2.0  
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/makepkg/sl1-prusa-slicer-git/src/PrusaSlicer/build
~~~~

and now?

greetings

sl1pkn07, 2021-09-24:
well. i have tested a lot of things, and i think i have found the gulty

the systemwide C/CXXFLAGS set up `-march=native` as default compiler flag. then i've change in build time to `-march=x86-64 -mtune=generic`, and the ctest is passed OK without problem

any dev can take a look why fail the 2rd test (and some time get segfault when use the program normally, like posted in https://github.com/prusa3d/PrusaSlicer/issues/4141#issue-608491280) and if is fixeable without change that flag?

greetings