prusa3d/PrusaSlicer #4143: Error in model slicing

ah1102, 2020-04-28:
### Version
2.2.0 ; 2.1.1
### Operating system type + version
macOS 10.15 Catalina

After slicing, the model breaks. In version 2.1.1, the same thing. In the cura slicer, the model is sliced normally.
File: Tail_Set.stl https://www.thingiverse.com/thing:2529936/files

[image]
[image]
[image]

rtyr, 2020-04-28:
The model is broken. You can clearly see on your screenshot, that it isn't sliced properly with Cura. Unfortunatelly it is too severe to repair it properly with Netfabb service. The only issue on the PrusaSlicer side is that it should recognize this model as non-manifold.

Netfabb
[image]

Simplify3D
[image]

Cura - Prepare
[image]

Cura - Sliced
[image]

ah1102, 2020-04-28:
Thank you for your help)

ah1102, 2020-04-28:
Here, on the forum, a woman somehow got the opportunity to print this beaten file https://forum.prusaprinters.org/forum/prusaslicer/error-in-model-slicing/#post-211382. She claims that she changed the diameter from 0.4 mm to 0.25 mm. But I checked and it does not play any role. What can be included in its settings so that artifacts disappear?

rtyr, 2020-04-28:
You mean this? It is sliced for Prusa SL1, which is SLA printer :-). 

[image]

ah1102, 2020-04-28:
Yes, but this is PrusaSlicer! Just wondering why her model was presented as in Cura?

rtyr, 2020-04-28:
SLA slicing is completely different from FDM slicing. 

And as I mentioned before, Cura didn't slice it properly.

ah1102, 2020-04-28:
I agree with you that the file is damaged, and in Cura it also displayed with errors, but with a smaller number.

You think this is just a PrusaSlicer slicing algorithm for FDM and SLA. I understood. Give thanks)

ah1102, 2020-04-29:
Another problem with hand files: Hand.stl https://www.thingiverse.com/thing:2529936/files

[image]
[image]

rtyr, 2020-04-29:
Same reason. But this can be fixed. Right click on the model and select "Fix through the Netfabb".

closed 2026-09-29 by prusaslicer-bot[bot]