prusa3d/PrusaSlicer #4144: Path for Python post-processing scripts gives only errors

Tillema, 2020-04-28:
### Version
PrusaSlicer 2.2.0 (Dutch)

### Operating system type + version
Win10

### Behavior
When I want to run a python script after exporting the G-code, I can set the path in the output options in print settings.
I checked several sources, but none of them gives answer to my specific question.
I want to set the path to the Python file here: **C:\Program Files\PrusaSlicer\resources\script.py**
The only thing that works was:
**C:\Users\simon\AppData\Local\Programs\Python\Python38\python.exe C:\Users\simon\Documents\script.py;**
But in that case, the script is in the wrong folder.
When I replace the last part by **C:\Program Files\PrusaSlicer\resources\script.py**, I get an error.
Tried a lot, but am not an expert on this field.
You can simply check if the program works by running the following Python code:

```
import sys
import time
print ("Hello World")
time.sleep(1)
```

Place it in the C:\Program Files\PrusaSlicer\resources folder and try to run it. The only thing I need is how to set the path correctly.

bubnikv, 2020-04-29:
Please check whether this helps:
https://github.com/prusa3d/PrusaSlicer/issues/3335#issuecomment-602246204

Tillema, 2020-04-29:
Got it. For me, placing the path (C:\Program Files\PrusaSlicer\resources\profiles\script.py) within quotation marks worked. like:
**C:\Users\simon\AppData\Local\Programs\Python\Python38\python.exe "C:\Users\simon\Documents\script.py";**
Thank you!

closed 2020-04-29 by Tillema

docD3D, 2024-04-06:
es increible que prusaslicer sea tan complicado para agregar una miniatura ... cura lo hace en 3 pasos... todavia no puedo hacerlo funcionar en  prusaslicer

emguepo, 2024-04-26:
> es increible que prusaslicer sea tan complicado para agregar una miniatura ... cura lo hace en 3 pasos... todavia no puedo hacerlo funcionar en prusaslicer

Tampoco he podido, a  mi me bota este error, tanto en prusa, como en orca  
[image]
           
[image]