prusa3d/PrusaSlicer #4145: Extra layer added during slicing of stacked models

jwg3, 2020-04-29:
Prusaslicer 2.2+
OS: MacOS 10.12.6 or Windows 10
Hardware: Mac or PC, any processor
3D Printer: Any, Marlin, RRF, etc.

I have created a model that stacks 10 medical shield visors using a 0.5mm separation-layer of 0.5mm columns between visors. The 10-stack model was made from two 5-stack models. The 5-stack model slices and prints with no issue. In the 10-stack the 5th, 6th, and 7th separation-layers have an extra print layer added by way of the 0.5mm separation-column being extended one layer. Subsequent printing fails. 

See attached jpeg image from Prusaslicer, PS config, and 10-stack STL model.

Changing slicing layer height changes which separation-layers are affected, but the slicing errors persist.

Changing the separation-layer distance in the model does not fix the problem.

[image]

[PSconfig.txt](https://github.com/prusa3d/PrusaSlicer/files/4549606/PSconfig.txt)

[PS 10x Visor Stack.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4549621/PS.10x.Visor.Stack.stl.zip)

JohnnyDeer, 2020-04-29:
Hi, can you please attach 3mf, rather than separate stl and config. Thank you

jwg3, 2020-04-29:
Done.

FidelCapo, 2020-10-15:
This issue should be fixed in next release.
Closing.

closed 2020-10-15 by FidelCapo