prusa3d/PrusaSlicer #4148: [Enhancement] Fix that darn scale (key) in the preview window

Photogad, 2020-04-29:
### Version
Any
### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Any
### Behavior

The darn scale (key) in the upper left corner is never accurate or really shows you what you really need to know.

This has been this way in Slic3r forever, I'm not sure if it can be fixed, but just asking in case it can. It seems like it would be a MAJOR quality of life improvement. The preview scale (key) in the upper left corner not being accurate has always bothered me.

In below example, I'm looking at layer height. It's same 0.20mm layer height for all the layers.

So why does the preview scale go from 0.000 to 0.390? Lol. :) It doesn't even show the exact color for the only layer height there (0.200). Makes no sense at all.

(this applies to all the preview modes, not just layer height)

[image]

_Is this a new feature request?_
YES

bubnikv, 2020-04-29:
You mean you want these legend thresholds to be aligned to less numeric digits?

Photogad, 2020-04-29:
> You mean you want these legend thresholds to be aligned to less numeric digits?

I guess what i'm saying is, simplify the legend so that it matches what is actually displayed, where possible.

So like in the above example photo, I'm looking at the layer height preview -- there's only actually 1 layer height present (0.20mm).  So why not just show the legend as 1 item and color for that, instead of a whole range that doesn't match at all.

Or if we had 2 layer heights, let's say a 0.20mm first layer and rest of layers 0.30mm. Then we show two colors and two legend items that match.

Same thing can apply for things like line width preview (where you're probably only going to have 1 to 5 different line widths for an entire print) or fan speed (auto cooling aside, PLA usually runs at 100% fan speed the whole print).

I understand some of the previews will have too many to show, maybe like print speed, and in that situation it makes more sense to have a range of items on the legend like present.  But I think it could easily be modified with an "if" statement or something like that if there's more than 5 items to be displayed, show it as a range instead of just a static value on the legend.

hope you understand.

Photogad, 2020-04-29:
i'll post more examples when i get back home to show what i mean.

Photogad, 2020-04-29:
Here are some example.

### 1 - In this example, it is correct how I would like all of the preview to be. We have items specifically on the preview key with their own color and clearly defined. I know these are named items and not numbers, but I am just using this as an example for items to follow.

### 2 - In this preview of layer height, the preview key have many colors that don't exist in the print. There are only TWO layer height used in the whole print. And none of the actual layer heights are shown on the key!! I am colorblind and looking at this, I would say that the yellow color is 0.217mm layers, but in fact I know it's 0.200mm layers!! The scale lies.

### 3 - Again, with line width. There are only two line widths used, but the whole scale and colors are shown. And again, I know the darker color is supposed to be 0.450 line width, but there is no 0.450 on the key! The closest number on the key is 0.454mm, but this is wrong and misleading. It lies.

### 4 - Fan speed. Only two fan speeds are used for this print. All the other colors and numbers on the scale key are irrelevant. At least with this one, the numbers on the scale key are correct... there is a 0 and 100. I know the two fan speeds are 0% and 100%.

### 5 - This is perfect!! Only one extruder is used here, only one color, and only one color on the scale.

[image]
[image]
[image]
[image]
[image]

Photogad, 2020-04-29:
`Even if it's not possible to get rid of all the extra unused items or colors on the scale key in upper left corner, can at least it be made so the real numbers are on there??`

rtyr, 2021-02-16:
Fixed in 2.3.0. Closing.

closed 2021-02-16 by rtyr