prusa3d/PrusaSlicer #4150: Add purge bucket support to Object setting to modify wipe options

KlungelSmurf, 2020-04-29:
### Version
2.2.0 Linux X64

### Operating system type + version
Linux Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
Mendelmax (old thingy)
Up mini
Prusa MK3s MMU2s (soon to be build)

### Behavior
New feature request
* I would like to use 3 options for purge material
  - waste tower
  - purge to infill
  - purge bucket

* By being able to purge ti infill and purge bucket you will reduce waste and save the environment.
  Also the print will be faster as you do not have to build up a tower keeping up with your project.

* On a tool change Slic3R should build G-code to move to the purge bucket and do the change. Then clearing out 'old' material should go to infill as much as possible. Then the (if there is any) remainder should go to bucket again. Then printing should continue.

* The retractable purge bucket from BigBrain3D is in a proper state of build and can be mounted on quite a lot of printers already. Also Prusa had a working model of the waste bucket (although it seems they have stopped working on it due to the volume of the purged material and because BigBrain3D had a good working model already).

* _Actual Results_

* _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
Yes

General considerations:

> Josef Prusa about waste buckets:

https://github.com/prusa3d/PrusaSlicer/issues/2617#issuecomment-512195936

Waste buckets are terrible. Waste tower makes the excess filament compact, but the same amount of material in bucket can have upto 10x the volume. You need to clean it multiple times during the printing. We started with waste bucket btw.

> BigBrain3D solved this issue with a thumbnail. ;-)
https://www.bigbrain3d.com/product/retractable-purge-mechanism/

Any waste mechanism involves moving the head to the side and do something like a toolchange and extrusion.

JohnnyDeer, 2020-04-30:
Hi @KlungelSmurf , did you tried "no spare layers" option in wipe tower settings? Its good way to reduce waste and save the environment.

KlungelSmurf, 2020-04-30:
@JohnnyDeer 
Euhhh, it says 'soon to be build' above right? ;-)
For now I am trying to learn C++ to do the request myself, but I think it is way over my head.
That does not stop me from trying though. :-)

Thank you for your reply. (Y)

closed 2026-09-29 by prusaslicer-bot[bot]