prusa3d/PrusaSlicer #4152: Version 2.2.0

Lanzentraeger, 2020-04-29:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-04-29:
Please provide 3MF project file.

carver0510, 2020-04-29:
Attached the 3mf file as requested.

[Fanduct.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4555354/Fanduct.3mf.zip)

rtyr, 2020-04-30:
Fan speed override works if bridging fan speed is set to higher value than regular fan speed. I believe it is intended behavior. For some reason, you have bridging fan speed set to lower value, which doesn't make much sense to me.

I believe there is no difference between 2.1.0 and 2.2.0 in this regard.

[image]

Lanzentraeger, 2020-04-30:
It makes Sense. The Fan is very effective, so if it would blow more, the bridging Layser will be blown down, so you will get a hanging bridge. With the Settings on Speed i have, its not Problem, to bridge over 100mm in a straith Line. The Settings are working perfect on 2.1.0 ... so why it is changed? My absolute minimus Speed is 30%, max 80% and Bridge 45%. If i print in Vase Mode, i need much Fan Speed to print with high Speed... so it makes Sense. It makes no Sense to me, to overwrite my Settings in this Version, so i have to change the Settings on every other Part i print? NO thanks..... There are more Problems on 2.2.0 ... also on the Tree Frog. If one Layer is ended on a small front Feet, the next Layer starts on this Feet, too. So you get 2 Layers without cooling Time on that small Feet, it also makes no Sense to me and is no Problem on 2.1.0 The sequence is not really working good on 2.2.0

rtyr, 2020-04-30:
I generated gcode from your 3MF project from both 2.1.0 and 2.2.0. Fan speeds are identical in the gcode.

Lanzentraeger, 2020-04-30:
So then please tell me, why the Slicer is accepting and doing my slow Fan Speed on Briding, when i set the max Fan Speed to 100%, but when i set max to example 80%, it is not working and Briding Fan Speed uses the max Speed then? Come on... when i tell the Slicer to bridge with 50, it hast to do a Bridge with 50. Not with 50 when i give max a 100 and 80 when i give max a 80. Thats makes no Sense, sorry

rtyr, 2020-04-30:
Sorry, I am not familiar with the history behind this parameter. There is no difference in fan speeds between 2.1.0 and 2.2.0 with your 3MF file. I am also unable to reproduce the behavior you are describing, only thing that matters when I am trying it is whether bridging fan speed is higher than regular fan speed. It was somehow a whole point of this parameter - to use higher fan speed for bridges, see https://manual.slic3r.org/expert-mode/cooling). But maybe you are right about the fact, that it should/can simply use specified speed regardless of regular fan speed.

Anyway, I will leave this for PS dev team to check and decide.

closed 2026-09-29 by prusaslicer-bot[bot]