prusa3d/PrusaSlicer #4153: [feature request] incremental extrusion rate for first layer printed in spiral mode.

printerbuilder, 2020-04-29:
### Version
_Version 2.2.0_
_win64_

### Windows 10
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### LUTUM / 4
_this machines uses general gcode files made with prusaslicer #s_

### Behavior
I am very (and happily impressed with the quality of the build and the stable gcode generated by prusa slicer! 

I still have a request, its related to spiral printing (very common in the clay printing scene)

When I select spiral but first print several bottom layers I get a large bulge on the outside of the print when the printer starts printing the 'spiral' part after the base layers.
Is it possible to set an incremental value for the first spiral layer? say start with 30% extrusion at the start of the first spiral layer always ending at 100% after completing the first spiral layer?

I have noted that no slicer currently available has that kind of feature. Is it very difficult to implement something like that as an option?

* _select spiral_
* _perfect incremental extrusion_
* _bulge at the start of the first spiral extrusion_

_Is this a new feature request?_ YES

[image]

about the picture;

this is a clay print, 
layer height is 1mm, printed with a 3mm nozzle.
The bulge is where the spiral movement starts, here Z displacement in regard to the layer below is zero but extrusion is 100%.
Once that first spiral layer is printed and the nozzle returns to the starting point Z has gradually moved up 1mm.
All following layers print perfect.

The rough white bit under the print is the printbed, an old slab of plaster.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-02-17:
This should be fixed since 2.3.0-beta. Closing.

printerbuilder, 2021-02-18:
Hi @rtyr

I just tried with the latest 2.3.0 version (updated today) I still get this bulge between the first solid layers and where the actual vase mode section starts printing.
Am I doing something wrong in the settings? 
Do I need to check some box other than the 'Spiral vase' box?
Or are there limits on the incremental distance?

Thank you for your feedback.

rtyr, 2021-02-18:
Please share your 3MF project file.

printerbuilder, 2021-02-18:
When i check slicing results with volumentric flow rates, there is no incremental path shown.
Just a line at a very low flow rates running from the center of the final bottom layer to the edge. After that it is full extrusion all the way around the pirimeter.

[disk-spiral-test.zip](https://github.com/prusa3d/PrusaSlicer/files/6002233/disk-spiral-test.zip)

printerbuilder, 2021-02-18:
the file is just a simple disk

printerbuilder, 2021-02-18:
[image]

bubnikv, 2021-02-19:
It only works in extruder relative mode.

printerbuilder, 2021-02-19:
Yes! Thank you, i got it working now.

> It only works in extruder relative mode. čt 18. 2. 2021 v 11:58 odesílatel printerbuilder  .

closed 2021-03-04 by rtyr