prusa3d/PrusaSlicer #4154: [Feature request] Retract settings Change at layer height

SphaeroX, 2020-04-29:
Slic3er PE offers great possibilities to change settings from a certain height. The color, meanwhile the GCode (which is very cool!).

But in my opinion there is still one option missing, which could take a lot of work out of it if you calibrate a new filament and **change the retract setting but at a certain height**.

So you could load three towers on the printing bed and change the retract settings every 5mm. Then start the print once and you can read off which setting is the best.

P-i-r-o, 2020-04-30:
There is a simple workaround.
If you use firmware retraction then you can change it with gcode. If you don't then try use it :) I recently wrote the script that adds desired gcode in desired intervals.
Here is this script adapted to retraction:

{if (layer_z> = filament_cost [0] * ((layer_z + z_offset - layer_z% filament_cost [0]) / filament_cost [0])) && (layer_z <(filament_cost [0] * ((layer_z + z_offset - layer_z% filament_cost [0]) / filament_cost [0]) + layer_height))} M207 S{retract_length [0] + (filament_density [0] * ((layer_z - layer_z% filament_cost [0]) / filament_cost [0]))} {endif }

Copy it to before or after layer change gcode section, change filament cost to tower segment height and filament density to retraction length increment.

c0m3d1an, 2020-08-10:
The above code throws an error with prusaslicer,

>  Please inspect the file \AppData\Local\Temp\.20736.gcode.tmp for error messages enclosed between
> !!!!! Failed to process the custom G-code template ...
> and
> !!!!! End of an error report for the custom G-code template ...

I'm assuming because of the percent operators?  I'm not a gcode expert, but I thought the only "proper" use of percent operators in gcode were to indicate the beginning and the end of the code...

P-i-r-o, 2020-08-10:
Oops! My fault. I accidentally putted '  ' between '>=' at the beginning. I don't know how it happened :/

Try corrected version:
{if (layer_z >= filament_cost[0] * ((layer_z + z_offset - layer_z % filament_cost [0]) / filament_cost[0])) && (layer_z < (filament_cost[0] * ((layer_z + z_offset - layer_z% filament_cost [0]) / filament_cost[0]) + layer_height))}M207 S{retract_length[0] + (filament_density[0] * ((layer_z - layer_z % filament_cost[0]) / filament_cost[0]))} {endif}

Here '%' is used as the modulo operator, not to indicate start/end of the code

c0m3d1an, 2020-08-10:
Oh awesome, thank you!  That fixed it!  As I said, I am by no means a gcode expert, so I am glad you were able to fix the code!

rtyr, 2021-02-16:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3635. Closing.

closed 2021-02-16 by rtyr