prusa3d/PrusaSlicer #4155: Perspective setting not saved

henrikkoch, 2020-04-29:
### Version
2.2.0+win64-202003211145

### Operating system type + version
Win10 pro

### 3D printer brand / version + firmware version (if known)
n/a

### Behavior
Every time I open this release slicer I have to select perspektive view. This setting is not being saved in between use of Slicer

If not the intention please change this for future versions

[image]

wjsteele, 2020-05-20:
Oh yes, please!  This is easily my number one complaint with PrusaSlicer.

ondratu, 2020-09-14:
I have same problem on Linux (2.2.0+linux-x64), sometimes it was load successful, sometimes not.

closed 2026-09-29 by prusaslicer-bot[bot]