prusa3d/PrusaSlicer #4158: Wrong Support-Calculation in PrusaSlicer v2.2.0

FlintEastwood, 2020-04-29:
### Version
2.2.0+ Win64

### Operating system type + version
Win10 64bit Home 1903

### 3D printer brand / version + firmware version (if known)
TwoTrees Sapphire-S / LerdgeX 3.0.5

### Behavior
Wrong Support-Calculation in v2.2.0
Support exeeds the build plate for unknown reason.
Adding a support-blocker doesn't help.
In v2.1.0 it works fine and support is calculated correctly.

[image]

#### Project File (.3MF) where problem occurs
[Kettenschutz_P3.zip](https://github.com/prusa3d/PrusaSlicer/files/4554922/Kettenschutz_P3.zip)

JohnnyDeer, 2020-04-30:
Hi @FlintEastwood , this is model issue. Fixing through netfabb will solve it.

FlintEastwood, 2020-04-30:
Hi @JohnnyDeer ,
I can't find anything to fix in netfabb. I did the model by myself in Blender really carefully. 
The model is closed, no flipped triangles, no unreferenced vertices/edges or triangles.

If there really is a problem in my mesh, can you show me please?
And why does it work with v2.10 ?

My best regards

JohnnyDeer, 2020-04-30:
I found just small part with intersecting faces, nothing what can explain support behavior, but fixing model truly helps.