prusa3d/PrusaSlicer #4159: [Request] Include "Add solid layer" option to Preview Mode "Add code..." contextual menu

fiveangle, 2020-04-30:
### Version
_2.2.0 win64_

I've observed similar issue to #3085 as seen below where bridging infill incorrectly attempts to bridge between 1 anchor while leaving the other end unsupported.  While that is being investigated, a really useful feature would be to add a "**_Add solid layer_**" option to the contextual menu that is available in _Preview Mode_ by right-clicking the `(+)` option at the current layer widget indicator:

[image]

This would allow users to add a solid layer indiscriminately where they desire, without having to due so using far more convoluted ways.

Thanks for considering this feature !

-=dave

#### Project File (.3MF) where problem occurs
[tamper_new.zip](https://github.com/prusa3d/PrusaSlicer/files/4555502/tamper_new.zip)

fiveangle, 2020-12-23:
Or at least fix this major problem of printing bridges to nowhere... what good does ironing do when basic print features aren't fixed ? 😿 

[image]

lukasmatena, 2020-12-23:
@fiveangle I'll not promise anything, but this issue has quite high priority.

closed 2026-09-29 by prusaslicer-bot[bot]