prusa3d/PrusaSlicer #4161: Compiled master and receive error

kevincraig, 2020-04-30:
### Version
version_2.2.0-256-g2875bc685

### Operating system type + version
MacOS 10.15.4

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s

### Behavior
* I followed the doc for [How to build - Mac](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md) and receive and Error Could not load bitmap: bullet_white

I ran the application with --loglevel 9 and got the following logs:
`22:07:20: Application built with libpng-1.4.12 but running with 1.6.35
22:07:20: Couldn't load a PNG image - file is corrupted or not enough memory.
22:07:20: Failed to load image from file "/Users/kcraig/Code/Personal/PrusaSlicer/build/src/Debug/../Resources/icons/PrusaSlicer_128px.png".
22:07:20: no bitmap handler for type 15 defined.
22:07:20: Application built with libpng-1.4.12 but running with 1.6.35
22:07:20: Couldn't load a PNG image - file is corrupted or not enough memory.
22:07:20: Failed to load image from file "/Users/kcraig/Code/Personal/PrusaSlicer/build/src/Debug/../Resources/icons/bullet_white.png".
22:07:20: Internal error: Could not load bitmap: bullet_white
`

* _Steps needed to reproduce the problem_
  run ./prusa-slicer from /PrusaSlicer/build/src
* _Expected Results_
  Able to run Prusa Slicer
* _Actual Results_
`[2020-04-29 22:07:20.497533] [0x000000010f05adc0] [error]   Uncaught exception: Could not load bitmap: bullet_white
Unhandled unknown exception; terminating the application.
22:07:20: Debug: Failed to load embedded PNG image for "wxART_ERROR"
22:07:20: Debug: Failed to load embedded PNG image for "wxART_ERROR"`

* _Screenshots 

[image]

Hopefully this is a simple fix and I just missed something in the setup.  I'm going to also pull down and build the 2.2.0 Release code and try and build it as well.

tamasmeszaros, 2020-04-30:
Hello, which commit sha have you built?

tamasmeszaros, 2020-04-30:
Hello,
could you please try to configure the build with cmake in a fresh directory and post the output?

kevincraig, 2020-04-30:
> Hello,
> could you please try to configure the build with cmake in a fresh directory and post the output? No need to build it, just the cmake configure output.

@tamasmeszaros  let me know if you need more than this.  I'm new to CMake so I wasn't sure if this was everything you needed.   

```~/Code/Personal/PrusaSlicer/deps/newBuild   master  cmake ..                                                           ✔  3 ⚙  8262  08:23:18
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- PrusaSlicer deps DESTDIR: /Users/kevincraig/Code/Personal/PrusaSlicer/deps/newBuild/destdir
-- PrusaSlicer deps debug build: ON
-- Found Git: /usr/bin/git (found version "2.24.1 (Apple Git-126)")
OS X SDK Path: /Applications/XCode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk
OS X Deployment Target (inferred from SDK): 10.15
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/kevincraig/Code/Personal/PrusaSlicer/deps/newBuild```

tamasmeszaros, 2020-04-30:
hi @kevincraig, it would help more if I could see the cmake output from configuring the PrusaSlicer project. I suspect this is the output from configuring the dependencies (deps subfolder).

kevincraig, 2020-05-04:
I just wanted to update this and let you know I still haven't been able to compile.  I've cleared the build folders several times and built command line and Xcode builds and both fail at the same place.  I'm working on getting an environment set up on a windows machine but I did have issues with that earlier as well but they were different so I'm digging through that tonight.

kevincraig, 2020-05-06:
@tamasmeszaros Just checking in to see if that log helped and if you need anything else from me.  I did continue with my Windows setup and I'm receiving linker issues there.  So I'm just focusing on the mac environment.  Is it because I'm on 10.15?

tamasmeszaros, 2020-05-18:
@kevincraig From the output it seems that nothing is linking to libpng dynamically which is ok. I can't tell what is going on. Maybe you can try with the latest master on a clean build environment. I'll leave this issue open and in case we bump into this, I'll post the fix.

kevincraig, 2020-05-18:
> @kevincraig From the output it seems that nothing is linking to libpng dynamically which is ok. I can't tell what is going on. Maybe you can try with the latest master on a clean build environment. I'll leave this issue open and in case we bump into this, I'll post the fix.

I will pull the latest master into a folder off root and make sure it's not a directory length thing as well.  I saw that it wasn't linking that library as well.  I've tried a few things around that.  I'll let you know what I find with the latest commit to master.  Thanks for the response.

kevincraig, 2020-05-18:
@tamasmeszaros Quick update just pulled from master into a clean folder and am getting the same issue running /src/prusa-slicer from terminal.  I'll keep digging as well.

bgiot, 2020-05-21:
Got the same issue with today 'pull' from repo.  Uncaught exception: Could not load bitmap: bullet_white

kevincraig, 2020-05-21:
> Got the same issue with today 'pull' from repo. Uncaught exception: Could not load bitmap: bullet_white
@bgiot 
Glad someone else is getting the same issue, maybe I'm not crazy.  hahaha If you come up with anything please let me know.  I haven't had a chance to dig into the logs that deep with my day job projects.  Wrapping up one this week so hopefully, I can look this weekend deeper.

lehighkid, 2020-05-21:
Pulled and compiled this morning on Mac OS X w/ same issue.

`Internal error: Could not load bitmap: bullet_white`

kevincraig, 2020-05-27:
I just pulled the latest from master and am still get the same error.

bgiot, 2020-06-18:
I tried again today with a fresh pull and same issue... any fix foresee ?

brettvitaz, 2020-06-20:
I am getting this as well.

lehighkid, 2020-06-21:
I was able to finally compile and run the latest pull successfully thanks to [this](https://github.com/torch/image/issues/137#issuecomment-174321330).  In my particular case I had Mono framework installed which was causing it to be built w/ an older version of libpng.  Removing that framework, clearing out the deps/build and /build dirs and recompiling worked.

bgiot, 2020-06-21:
@lehighkid  Not an option for many people...

brettvitaz, 2020-06-21:
This comment from @dirkessl resolved the issue for me: https://github.com/prusa3d/PrusaSlicer/issues/4407#issuecomment-647015714

> change this in PrusaSlicer-master/deps/wxWidgets/wxWidgets.cmake:
> -DwxUSE_LIBPNG=builtin
> and rebuild deps and compile again.

kevincraig, 2020-06-22:
I can confirm that this fix from @dirkessl @brettvitaz did allow me to compile and run the code.

> This comment from @dirkessl resolved the issue for me: [#4407 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/4407#issuecomment-647015714)
> 
> > change this in PrusaSlicer-master/deps/wxWidgets/wxWidgets.cmake:
> > -DwxUSE_LIBPNG=builtin
> > and rebuild deps and compile again.

bgiot, 2020-06-22:
The solution provided by @brettvitaz fixed also the issue on my side. Can someone check on Windows if it works also ?

bgiot, 2020-06-22:
I created the PR #4424

tamasmeszaros, 2020-06-26:
Guys, as a quick diagnostic, could someone suffering from this issue run the following script:
``` python
import subprocess
import sys

def otool(s):
    o = subprocess.Popen(['/usr/bin/otool', '-L', s], stdout=subprocess.PIPE)
    for l in o.stdout:
        if l[0] == '\t':
            yield l.split(' ', 1)[0][1:]

if len(sys.argv) < 2:
    exit(-1)

need = set([sys.argv[1]])
done = set()

while need:
    needed = set(need)
    need = set()
    for f in needed:
        need.update(otool(f))
    done.update(needed)
    need.difference_update(done)

for f in sorted(done):
    print f
```

And run it with python: `python findlibs.py ./prusa-slicer | grep png`

The output should be empty string. If not, please post the output you get.

kevincraig, 2020-06-26:
> Guys, as a quick diagnostic, could someone suffering from this issue run the following script:
>
> ```python
> import subprocess
> import sys
>
> def otool(s):
>     o = subprocess.Popen(['/usr/bin/otool', [...]

I ran this today on a fresh clone of the master branch and got no output and still with a fresh build can't run ./prusa-slicer

tamasmeszaros, 2020-06-30:
I've been digging into this, and from @kevincraig 's configure output `-- Found PNG: /Users/kevincraig/Code/Personal/PrusaSlicer/deps/build/destdir/usr/local/lib/libpng.a (found version "1.4.12")` there is something wrong with this, it should be version 1.6.35 and I'm getting it correct when I configure PrusaSlicer with the newly built deps.

This is so weird, if you have built libpng as part of the dependency build scripts, it should not be 1.4.12, the scripts always pointed to 1.6.35.
So @kevincraig, you could try to remove all the files in your `Users/kevincraig/Code/Personal/PrusaSlicer/deps/build/destdir/usr/local/` and start over with building the dependencies. I can't come up with anything better at the moment.

tamasmeszaros, 2020-07-13:
Guys, please check if you can build with the latest master. Please delete all the previous build artifacts including the dependencies and start from scratch.

kevincraig, 2020-07-13:
> Guys, please check with the latest master. Delete all the previous build artifacts including the dependencies and start from scratch.

I pulled down a fresh copy of the master branch and I was able to compile and run the app.

tamasmeszaros, 2020-07-13:
Many thanks to @bgiot

closed 2020-08-19 by tamasmeszaros