prusa3d/PrusaSlicer #4163: Unable to print one at time, Some object are too tall and cannot be printed (not true)

marcottt, 2020-04-30:
### Version
2.2.0

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
Anycubic kossel linear plus

### Behavior
I'm trying to print 4 object, one at time but cannot slice because i get error "some object are too small".  object are small (tallest is 50mm)  and i tried also to change the clearance value from 400 to 0 many value, no change

I printed everything with cura, no problem so printer can do it... i think there is some problem with clearance or gauntlet size but i cannot find a way to manage it.

tried als with only 2 pieces, same issue.

#### Project File (.3MF) where problem occurs

[1588177649-140plarosso-alto-defintiivo_1.zip](https://github.com/prusa3d/PrusaSlicer/files/4557122/1588177649-140plarosso-alto-defintiivo_1.zip)

rtyr, 2020-04-30:
It is expected with your values. Set **height** to at least to 50mm. Closing.

[image]

closed 2020-04-30 by rtyr

marcottt, 2020-04-30:
thanks. i read wrong about height