prusa3d/PrusaSlicer #4167: Bridge layer is "feathery" extruded lines too far apart.

DavidPhillipOster, 2020-05-01:
### Version
PrusaSlicer PrusaSlicer-2.2.0+-2020003211132

### Operating system type + version
macOS 10.15.4 (19E287)

### 3D printer brand / version + firmware version (if known)
Stock Prusa i3 mk3 firmware=prusa3d_fw_3_8_0_MK3.zip 0.4mm nozzle. PLA.

### Behavior
* After slicing with the Prusa standard PLA profile the first bridging layer has gaps between the extruded lines. I believe this is because in all other cases the extruded lines are squished by being directly on top of build plate or previous layers.

* To reproduce
 Slice with standard PLA profile, 0.20mm SPEED setting.

* _Expected Results_

Just as PrusaSlicer has an advanced setting to slow extrusion when bridging, I'd like to have a setting to move the extruded lines closer together, or maybe over-extrude bridging lines.

* _Actual Results_
As shown in the attached photo, there are gaps between the extruded lines when bridging.

_Is this a new feature request?_
YES

Example

[image]

rtyr, 2020-05-01:
Look at this https://manual.slic3r.org/advanced/flow-math. You can find some info there how it works and why it is not a good idea to overlap bridging paths. You can experiment with bridge flow, bridge speed and cooling - those options are already available.

DavidPhillipOster, 2020-05-01:
Thank you. That link convinces me that improvement in this case is not possible. Closing this issue.

Thank you for teaching me. I'll either fill the fibrous result with white glue when done, or use supports instead of bridging.

closed 2020-05-01 by DavidPhillipOster