prusa3d/PrusaSlicer #4170: [Feature Request] "External perimeters first" on first layer

towe96, 2020-05-01:
### Version
2.2.0-win64

### Operating system type + version
Windows 10 1909

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
When slicing a model with a complicated first layer - for example models with small circles, the standard "internal perimeters" settings results in those circles being printed first with no support. This can easily lead to them lifting off the build plate.

Enabling the "external perimeters first" options allows those small features to adhere much better to the more robust, larger outer perimeters.
However, at higher layers, printing the internal perimeters first tends to improve quality on overhangs.

Being able to print only the first layer with "external perimeters first" activated would reduce the need for brims and rafts while still providing consistent first layer performance and good overhangs at higher layers.
[image]

guestisp, 2020-05-01:
Duplicate of #2846

guestisp, 2020-05-01:
Anyway, in this particular print, just print external perimeter first always enabled. Printing the internal first, helps on overhangs, here you don't have any overhang.

guestisp, 2020-05-01:
as workaround, reduce the perimeter number and print with infill 0%, leave an hole in that circle.

rtyr, 2020-05-01:
As guestisp mentioned it is a duplicate of #2846. Closing.

closed 2020-05-01 by rtyr

KimmoHop, 2020-08-07:
@towe96 - One way to do this: select ear saver and right click on it ->add modifier ->box). Resize the box to cover whole part. Then make modifier same height as first layer (unlock scale and size, set z-size e.g. 0.2) and drop to bed (small icon next to z-position). Then right-click on modifier box settings cog, add settings -> layers and perimeters -> check external perimeters first. Finally enable that and slice 👍 

[image]

[image]

In IdeaMaker one can use per-layer settings to enable print external shells first on Z 0.0 - 0.2mm. Therefore I thought there could be a way to do the same in PrusaSlicer as well! Since PS does not show moves within layer, I used IM to verify that the method above works. I haven't printed this yet, but first layer should be easier now.

[image]

Pro tip: add modifier before adding instances

phazei, 2024-03-27:
@KimmoHop This is exactly what I was looking for, for 90% of my print I need external layers first for dimensional accuracy, and for a couple layers I need it off due to overhangs.  I was hoping I could do that as a modifier.  I'm looking all over for that setting, still can't find it.  Has it been removed as a option for a modifier?  Or am I just missing it?